zlgopen/awtk #225: 字体切换

ytchw101, 2019-09-03:
程序资源中添加两个字体， 如  
assets_manager_add(rm, font_default);
assets_manager_add(rm, font_default_ttf);

怎么动态切换，？

xianjimli, 2019-09-04:
在style文件中指定，通过切换不同的style来切换字体。可用参考一下awtk-hello项目。
```
<label>
  <style name="big_green">
    <normal text_color="green" font_size="24" text_align_h="center" border_color="#a0a0a0"/>
  </style>
  <style name="led">
    <normal text_color="green" font_name="mono" font_size="18" text_align_h="center" border_color="#a0a0a0"/>
  </style>
  <style name="italic">
   <normal text_color="green" font_name="italic" font_size="32" text_align_h="center" border_color="#a0a0a0"/>
  </style>
  <style name="bold">
   <normal text_color="green" font_name="bold" font_size="32" text_align_h="center" border_color="#a0a0a0"/>
  </style>
</label>
```

ytchw101, 2019-09-04:
不能程序中动态切换吗？  我现在需求是： 添加几个.ttf矢量字体 如
  assets_manager_add(rm, font_cuhei);
  assets_manager_add(rm, font_default);

程序中根据选择的字体和字号去 解析想要解析的字符串，得到字模数据，  不同字体 得到字模数据肯定不一样。。。。。。想实现这个   

之前获得字模数据我是这样操作：
font_get_glyph(g_canvas->font, str, font_size, gy)；//得到字模数据在gy中
g_canvas为自定义全局变量， 即native_window_create_internal中增加 g_canvas = canvas_init(c, lcd, font_manager());

目前使用 widget_set_style_str(label, "font_name", "cuhei");方式可以实现界面字体变化， 主要是font_get_glyph中第一个font_t* 该怎么指定。。？

xianjimli, 2019-09-04:
可以通过font manager去获取font对象。

ytchw101, 2019-09-04:
可以了

closed 2019-09-04 by ytchw101