zlgopen/awtk #230: F03 看到程序是跑起来了，就是不显示

masterchen, 2019-09-17:


xianjimli, 2019-09-18:
103最终显示时会调用src/lcd/lcd_reg.inc中的write_data_func。

不同的板子的驱动有些不同，你在write_data_func设置断点调试一下。

masterchen, 2019-09-18:
我单步跟踪了一下， 跑到memset这个地方就飞了，这个地方应该是有Bug，断点看了一下size传进来是8196
ret_t tk_mem_init(void* buffer, uint32_t size) {
  return_value_if_fail(buffer != NULL && size > MIN_SIZE, RET_BAD_PARAMS);

  memset(buffer, 0x00, size);
  s_mem_info.buffer = (char*)buffer;

另外，我的触摸屏是AT42CXX 是IIC不是SPI， 所以想先禁用touch， 所以在初始化的时候去掉触摸的初始化，但是这会导致 AWT起来以后一致在获取touch数据 挂死在gettouchdata , AWTK 如何配置不使用TOUCH?

masterchen, 2019-09-18:
举步维艰啊， 单步发现

发现这个memset 也飞了，size的值是16640， 能说明一下什么原因么

lcd_t* lcd_reg_create(wh_t w, wh_t h) {
  static lcd_reg_t reg;

  lcd_t* lcd = (lcd_t*)(&reg);
  uint32_t size = sizeof(reg);
  system_info_t* info = system_info();
  return_value_if_fail(lcd !

xianjimli, 2019-09-18:
看下stack设置的多大，感觉是stack溢出了。参考一下： https://github.com/zlgopen/awtk/blob/master/docs/debug_tips.md

masterchen, 2019-09-18:
设置了一下， 没有在这个地方出问题了，但是还是跑不起来， 不知道什么原因。我的板子是discovery的，已经无力了，😔

1kkiot, 2019-09-20:
..\awtk\src\tkc\value.c(620): error:  #120: return value type does not match the function type
    return &(v->value.sized_str);
..\awtk\src\tkc\value.c(636): error:  #120: return value type does not match the function type
    return &(v->value.binary_data);
我的编译报错，请问 如何改正