zlgopen/awtk #231: Linux平台下测试HelloWorld-Demo时edit控件无法输入空格符问题

Kevin-zzw, 2019-09-18:
[image]

Kevin-zzw, 2019-09-18:
再补充几个测试下来认为可能是问题的现象，麻烦确认下
[image]
[image]
[image]

tangMans, 2019-09-18:
问题一：
【问题描述】关于不能输入空格
【问题分析】之前AWTK做了限制不能输入空格
【解决方案】麻烦更新最新的AWTK源码

问题二：
【问题描述】清空编辑框内容后没有边框以及输入第一个字符光标显示在左边
【问题分析】样式文件default.xml中的edit控件缺少了“empty_focus”样式
【解决方案】可以参考AWTK中的default.xml样式文件下edit控件的“empty_focus”，将其复制到 
                   HelloWorld-Demo中或者等待下一次HelloWorld-Demo代码的更新

Kevin-zzw, 2019-09-18:
问题一：刚更新了最新的源码，看到commit的代码里去掉了限制空格的问题，现在测试demo也正常了！
问题二：从awtk/demos/assets/raw/styles/default.xml中拷贝了edit控件的“empty_focus”样式到demo中，测试结果还是一样。
[image]

tangMans, 2019-09-19:
问题二：
将“empty_focus”样式拷贝过去后，需要执行HelloWorld-Demo下的 assets_gen.sh脚本重新生成资源，不知道这步是否执行了没？

Kevin-zzw, 2019-09-19:
问题二
执行assets_gen.sh后可以了。修改样式文件后我以为scons -c后再重新scons会全部重新编译一次。个人建议，看看后续有没有办法把scons跟sh结合起来，实现一步编译有改动的文件。

closed 2019-09-19 by Kevin-zzw