zlgopen/awtk #232: 怎么加载外部库文件（.so文件）和外部头文件 (我要移植到arm-linux平台)

YinWenshuo, 2019-09-20:
我自己写了一个程序，你就当它是类似于awtk-examples里面的某个例子吧，然后，我要用到以前写的一个程序库文件（.so文件）及一些头文件，那我要怎么把库文件和头文件编译进去？我要把它移植到arm-linux板子上。

谢谢！

xianjimli, 2019-09-20:
以awtk-hello为例，你把自己的库加到APP_LIBS里就行了：

```
APP_CCFLAGS = ''
APP_LIBS = ['assets']
APP_LIBPATH = [APP_LIB_DIR]

DefaultEnvironment(
  CPPPATH   = awtk.CPPPATH,
  LINKFLAGS = awtk.LINKFLAGS,
  LIBS      = APP_LIBS + awtk.LIBS,
  LIBPATH   = APP_LIBPATH + awtk.LIBPATH,
  CCFLAGS   = APP_CCFLAGS + awtk.CCFLAGS, 
  OS_SUBSYSTEM_CONSOLE=awtk.OS_SUBSYSTEM_CONSOLE,
  OS_SUBSYSTEM_WINDOWS=awtk.OS_SUBSYSTEM_WINDOWS)
```

YinWenshuo, 2019-09-20:
多谢回复，但还是编译报错：/usr/bin/ld: skipping incompatible lib/libBle.so when searching for -lBle
/usr/bin/ld: cannot find -lBle

是不是路径设置不对？请指教。

我现在是这么做的：
我的库文件是libBle.so,libBleSch.so我把他们放在awtk-hello/lib/目录下，然后修改sconstruct如下：
import os
import sys
import platform

sys.path.insert(0, '../awtk/')
import awtk_config as awtk

APP_ROOT = os.path.normpath(os.getcwd())
APP_BIN_DIR = os.path.join(APP_ROOT, 'bin')
APP_LIB_DIR = os.path.join(APP_ROOT, 'lib')

os.environ['APP_ROOT'] = APP_ROOT;
os.environ['BIN_DIR'] = APP_BIN_DIR;
os.environ['LIB_DIR'] = APP_LIB_DIR;

APP_CCFLAGS = ''
APP_LIBS = ['libBlu.so']
APP_LIBS =APP_LIBS + ['libBluSch.so']
APP_LIBPATH = [APP_LIB_DIR]

DefaultEnvironment(
CPPPATH = awtk.CPPPATH,
LINKFLAGS = awtk.LINKFLAGS,
LIBS = APP_LIBS + awtk.LIBS,
LIBPATH = APP_LIBPATH + awtk.LIBPATH,
CCFLAGS = APP_CCFLAGS + awtk.CCFLAGS,
OS_SUBSYSTEM_CONSOLE=awtk.OS_SUBSYSTEM_CONSOLE,
OS_SUBSYSTEM_WINDOWS=awtk.OS_SUBSYSTEM_WINDOWS)

SConscript(['src/SConscript'])

YinWenshuo, 2019-09-20:
网上查了下，/usr/bin/ld: skipping incompatible错误是库文件版本与平台版本不对应，我的这两个库（libBle.so & libBleSch.so）是使用某个arm平台的编译链编译的，那么，编译这个应用（例如awtk-hello）需要怎么设置下编译环境吗，也要使用同样的编译链编译？我要将这个应用移植到arm-linux平台上，那么如何设置编译环境来使用arm平台的编译链来编译这个应用？

xianjimli, 2019-09-20:
#for linux-fb
sys.path.insert(0, '../awtk-linux-fb/')

closed 2019-09-23 by YinWenshuo