zlgopen/awtk #237: 请教一下移植的arm版本颜色显示成这样，是怎么造成的？

xsjqqq123, 2019-09-25:
[image]
使用的是awtk-linux-fb。图中的字体看起来颜色有点花，正常应该是纯白的。

YinWenshuo, 2019-09-25:
我也遇到同样的问题。。。

xianjimli, 2019-09-25:
这是创建lcd对象用的函数与实际framebuffer格式不一致引起的。

AWTK中LCD命名时，按照内存字节地址从低到高命名的。如：lcd_mem_bgra8888_create_double_fb，那第一个像素来说，第1个字节是blue, 第2个字节是green，第3个字节是red，第4个字节是alpha。

xsjqqq123, 2019-09-25:
加了些打印，fb_is_bgra5551为true，这个判断貌似有点问题。直接修改成lcd_mem_rgb565_create_double_fb，又变成黄色的，

xsjqqq123, 2019-09-25:
有空再看下我们用的是什么格式 @_@

YinWenshuo, 2019-09-25:
在fb_setting函数里面
 ...
fb->var.bits_per_pixel = 32; 
  fb->var.transp.offset = 24;
  fb->var.red.offset = 16;
  fb->var.green.offset = 8;
  fb->var.blue.offset = 0;

  fb->var.transp.length = 8;
  fb->var.red.length = 8;
  fb->var.green.length = 8; 
  fb->var.blue.length = 8;
...

创建lcd时（在lcd_linux_create_flushable函数里面）
lcd = lcd_mem_bgra8888_create_double_fb(w, h, online_fb, fb->fbmem1);

以上的做法应该没错啊，但是还是有些字显示“花”了，有些字显示就ok

有没有相关的lcd创建的文档呢。

xianjimli, 2019-09-25:
写个简单的RGB颜色显示函数测试一下。

soyzhc, 2019-11-18:
你图片的颜色是反了, 正常应该是蓝色, 你的显示绿色,
应该是rgb的顺序和你的屏刚好反过来了