zlgopen/awtk #238: 以BITMAP_FMT_GRAY显示不了

ytchw101, 2019-09-25:
之前以BITMAP_FMT_BGR565显示图片可以，但感觉多占空间，  现在改成BITMAP_FMT_GRAY格式，显示不了？

代码：
1、BITMAP_FMT_BGR565格式：
主要
	bitmap_t* img = bitmap_create_ex(w,h,w*2,BITMAP_FMT_BGR565);
        ......
     	 ret_t res = bitmap_init(img,label->w,label->h,BITMAP_FMT_BGR565,pbmpdata);
	 res = canvas_draw_icon(c, img, label->w/2, label->h/2);

注：pbmpdata为像素点数据, 如只显示一个点则数据为 0xFF,0xFF
注:  awtk_config.h 中定义 #define WITH_BITMAP_BGR565  1

2、BITMAP_FMT_GRAY格式：
主要
	bitmap_t* img = bitmap_create_ex(w,h,w,BITMAP_FMT_GRAY);
        ......
     	 ret_t res = bitmap_init(img,label->w,label->h,BITMAP_FMT_GRAY,pbmpdata);
	 res = canvas_draw_icon(c, img, label->w/2, label->h/2);
     
注：pbmpdata为像素点数据, 如只显示一个点则数据为 0xFF
注:  awtk_config.h 中定义 #define WITH_BITMAP_GRAY  1    (不知可正确？)

xianjimli, 2019-09-25:
BITMAP_FMT_GRAY目前还没支持。

ytchw101, 2019-09-25:
哦 好吧

closed 2019-09-26 by ytchw101