zlgopen/awtk #243: 关于spin_box和slider控件，再请教两个问题

YinWenshuo, 2019-10-09:
1、spin_box的值怎么让他只显示小数点后两位？
<spin_box x="105" y="middle" w="70" h="100%" min="0.50" max="1.50" value="1.00" step="0.01" input_type="float" auto_fix="true" name="SpinBoxDist2d" />
一开始是显示1.00，我一操作spinbox增减，就显示成1.0100000了或者0.9900000了

2、slider的min、max、value、step值有什么要求吗，只能是0或者正整数？我下面的写法会不行，拖不动slider（我的目的是slider的值从-80.0到80.0）
<slider x="180" y="middle" w="-200" h="100%" min="-80.0" max="80.0" value="-10.0" step="1.0" name="SliderDist3dY"/>

xianjimli, 2019-10-09:
1.我这里没问题啊，在编辑里显示是0.99或1.01。至于你取出来放到别的地方，或者打印处理，就要看你怎么格式化的了。用printf格式化，你可以参这篇文章：https://blog.csdn.net/abcdu1/article/details/75090369

xianjimli, 2019-10-09:
2.slide的min/max/value/step目前是uint16_t类型的，也就是非负整数。如果有浮点数和负数的需求，我想一下，看能不能改成double类型。

YinWenshuo, 2019-10-09:
第一个问题，我再查一下，我是在别的地方做了处理，然后显示就多了好多0,

第二个问题，我写成min=0 max=160 然后用的时候再-80.0，转换一下。

多谢回复。

另外，问下awtk团队有没有微信交流群或者QQ交流群。

xianjimli, 2019-10-09:
微信群超过100人，只能邀请进群。你加我微信：xianjim 我拉你进群吧。

xianjimli, 2019-10-09:
我改成double类型了，这样方便一些，你有空试试。

closed 2019-10-09 by YinWenshuo