zlgopen/awtk #245: 请教几个问题

YinWenshuo, 2019-10-09:
一、 关于widget_set_text(label, L"hello awtk!");
1、第二个参数 L"hello awtk!"中的L是代表啥意思？
2、多语言翻译使用widget_set_tr_text，那感觉widget_set_text函数就可以不要了吧

二、我有个回调函数
int calibrationCallback(int calibrationRes)
{
    int status = calibrationRes;
    log_debug("\n\n\n##%s res=%d,%d\n\n\n",__func__,calibrationRes,status);
    return status;
}
我需要根据这个回调函数返回的calibrationRes值对某个界面文件里面的某个label设置不同的状态，我应该怎么做，在这个回调函数里面我没法获得界面文件的句柄，也就是ctx。

xianjimli, 2019-10-09:
L"hello awtk!" 是让编译器把"hello awtk!"从char*转换成wchar_t*的类型，也就是UNICODE。
不需要翻译的用widget_set_text，需要翻译的用widget_set_tr_text

xianjimli, 2019-10-09:
一般来说任何回调函数都应该有ctx，否则只能通过全局变量来传递数据。

xianjimli, 2019-10-09:
windon_manager()函数可以获取窗口管理器对象，再通过窗口管理器对象找到对应的窗口和控件。

YinWenshuo, 2019-10-09:
ok，多谢。我感觉我对awtk整个思路、框架可能理解不深刻、不深入。有没有视频教程，或者说比较深入浅出的指导性的、框架、思想层面的文档。

closed 2019-10-14 by YinWenshuo