zlgopen/awtk #246: 关于 资源放到SD卡或者QSPI FLASH中，使用文件系统如何操作，有无写好的文档 或者怎么操作步骤？

ytchw101, 2019-10-10:
如题。

xianjimli, 2019-10-10:
这个我没实际的经验，你可以参考Linux版本弄。

ytchw101, 2019-10-10:
是参照 移植示例中的 linux-framebuffer （嵌入式 Linux 请用本项目） 这个工程吗？

xianjimli, 2019-10-10:
大概思路是这样的：

- 1.实现文件系统的接口。参照：https://github.com/zlgopen/awtk/blob/master/src/platforms/pc/fs_os.c

- 2.把资源文件拷贝到flash中(可能要做成镜像文件整体下载，这个我不懂)。

- 3.编译时定义宏WITH_FS_RES。

- 4.初始化时指定资源文件的路径。

ytchw101, 2019-10-19:
按照步骤弄了下  没弄好！  最终每次在 tk_init里面的 return_value_if_fail(tk_init_internal() == RET_OK, RET_FAIL);  这句直接崩溃掉！  不知什么原因？   目前没把资源放在flash中，而是放在U盘里

xianjimli, 2019-10-19:
可以单步调试不？

ytchw101, 2019-10-21:
1.定位代码，运行到tk_init_internal里面的
  return_value_if_fail(input_method_set(input_method_create()) == RET_OK, RET_FAIL); 这一句 崩溃  ，如果注释掉这一句，不会崩溃。
  查看了之前的备份工程， （烧写awtk源码为在2019/08/14更新以及之前的工程，不会崩溃， awtk源码为在2019/09/02之后的工程，会出现崩溃， 目前没找到原因！）

2. 目前注释掉会引起崩溃的那一句，调试其他， 有个问题问下：就是	我现在把资源文件夹assets
整个拷贝到U盘里,   检测U盘插上时   U盘会被创建作为一个目录为"/UPAN"为目录的文件夹使用(目前只要U盘插上我对U盘这个目录下读写都正常) 那么：
tk_init(lcd_w, lcd_h, APP_MOBILE, NULL, "/UPAN");  最后一个参数，是直接传递这个根目录就可以吗?
加载资源：
比如
是 assets_manager_load(rm, ASSET_TYPE_STRINGS, "zh_CN");      还是
assets_manager_load(rm, ASSET_TYPE_STRINGS, "/UPAN/assets/inc/strings/zh_CN");  ？
都没加载成功。。

soyzhc, 2019-10-21:
要保证tk_init传入资源assets的父文件夹

ytchw101, 2019-10-21:
哦   那assets_manager_load就直接指定类型和名称就可以吗》？

ytchw101, 2019-10-22:
assets_manager_load 加载图片资源 失败。。。

因为之前不是放在文件系统里面， 图片资源和字体资源 我都是通过assets_gen生成的 .res文件， 那么请问 我现在资源放在文件系统里 还是加载这个.res文件吗？  还是资源不作处理 直接放原始文件 比如：
count.png   default.ttf   然后 加载时直接
assets_manager_load(rm, ASSET_TYPE_IMAGE, "count");   
assets_manager_load(rm, ASSET_TYPE_FONT, "default");   

2.图片或字体资源如何转成.data文件， 我用assets_gen转成的图片和font资源， 怎么都是转成.res文件  ？

ytchw101, 2019-10-23:
资源问题 搞定了 是我fs_os.c有个接口问题   现在就是那个return_value_if_fail(input_method_set(input_method_create()) == RET_OK, RET_FAIL); 这一句 崩溃问题  没解决？？

closed 2019-10-24 by ytchw101