zlgopen/awtk #248: 你好请教一个关于使用scroll_bar_m控件的问题

lizhenningi, 2019-10-10:
我在小屏LCD中使用scroll_bar_m控件，设置控件高度为11（即：h="11"）,但是scroll_bar_m控件中的滑块不会按比例缩小，滑块的显示会超出scroll_bar_m控件的高度。我能够通过调整XML文件来解决这个问题吗？

lizhenningi, 2019-10-10:
屏幕大小128*64
`<window style="default" move_focus_prev_key="up" move_focus_next_key="down" >
	<view name="statu_bar" x="0" y="0" w="128"  h="12" style="top_bar_bg" >
  		<image draw_type="icon" name="phone_mode" image="register_failed" x="0" y="0" w="12" h="100%" />
   		<label name="phone_number" x="12" y="0" w="60" h="100%" text="" style="top_text"/>
	</view >
	
	<view name="call_bg" x="0" y="12" w="100%" h="25" style="menu_text_lift" >
		<edit name="input_number" x="2" y="0" w="124" h="24" input_type="email" style="number_input" focus="true" />
		<image x="1" y="24" w="126" h="1" draw_type="icon" name="down_line" image="down_line"/>
	</view>
	<pages x="0" y="38" w="100%" h="13" name="page11" style="nostyle">
		 <list_view name="search_list_view" x="0"  y="0" w="100%" h="100%" style="nobg" item_height="13" >  
		  	<scroll_view name="search_block" x="0"  y="0" w="123" h="100%" style="left" >    
			  <button style="menu_into_item" name="search_res" enable="false" focusable="true">
          			<label style="white_left" x="2" w="48" h="100%" text="" />
          			<label style="white_left" x="58" w="65" h="100%" text=""/>
        	  </button>
		  	</scroll_view>
		  	<scroll_bar_m name="bar" x="right" y="0" w="4" h="100%" value="0"/>
	 	 </list_view>
	</pages>
  <view name="call_soft_key" x="0" y="bottom:0" w="128" h="12" style="softkey_bg" is_keyboard="true">
	<button style="nostyle" name="soft_key_1" x="0"  y="0" w="32" h="100%" focusable="false" tr_text="Dial"   is_keyboard="true"/>
    <button style="nostyle" name="soft_key_2" x="33" y="0" w="31" h="100%" focusable="false" tr_text="" 	  	is_keyboard="true"/>
    <button style="nostyle" name="soft_key_3" x="65" y="0" w="31" h="100%" focusable="false" tr_text="Delete" is_keyboard="true"/>
    <button style="nostyle" name="soft_key_4" x="97" y="0" w="31" h="100%" focusable="false" tr_text="Clear"  is_keyboard="true"/>
  </view>
</window>`

xianjimli, 2019-10-10:
好的，一会儿，我看看。

xianjimli, 2019-10-10:
你更新代码试试，看看是不是你想要的。

lizhenningi, 2019-10-10:
是我想要的效果，谢谢

closed 2019-11-05 by xianjimli