zlgopen/awtk #250: 关于spin_box和slider再请教两个问题

YinWenshuo, 2019-10-13:
1、slider在前几天，min、max、value、step的值改成了可以设置double了，但仍然不能是负数？
2、操作spin_box可以改变spin_box的值，显示也正确，但是，如果调用edit_set_int(spin_box,value);当value是两位数时，个位数不显示 了，例如，edit_set_int(spin_box,34);spin_box显示3
其实我是想仿照demo里面的color picker，来实现操作spin_box可以改变slider滑块位置，拖动slider可以改变spin_box的显示值。
我的xml文件如下：
<view >
   <label x="0" y="middle" w="100" h="100%" text="xoffset"/>
   <spin_box x="105" y="middle" w="80" h="100%" min="-40" max="40" value="0" step="1" input_type="int" auto_fix="true" name="F_SpinBoxXoffset" />
   <slider x="190" y="middle" w="-210" h="100%" min="-40" max="40" value="0" step="1" name="F_SliderXoffset"/>
</view>

我的c代码如下：
static ret_t on_F_XoffsetChanging(void* ctx, event_t* evt) {
  widget_t* win = (widget_t*)ctx;
  widget_t* target = WIDGET(evt->target);

  if(0==strcmp("F_SliderXoffset",target->name))
  {
    widget_t* spin_box = widget_lookup(win, "F_SpinBoxXoffset", TRUE);
    edit_set_int(spin_box,widget_get_value(target));
    log_debug("slider front xoffset value = %d \n",(widget_get_value(target)));
    sCalibUndistortPrms.xoffset=widget_get_value(target);
  }
  else if(0==strcmp("F_SpinBoxXoffset",target->name))
  {
    widget_t* slider = widget_lookup(win, "F_SliderXoffset", TRUE);
    widget_set_value(slider,(edit_get_int(target)+40));
    log_debug("spin box front xoffset %d \n",edit_get_int(target)+40);
    sCalibUndistortPrms.xoffset=edit_get_int(target)+40;
  }

  return RET_OK;
}

在注册按键事件函数里面有如下代码：
. . .
     else if (strstr(name, "F_SpinBoxXoffset"))
        {
            widget_t* spin_box = widget_lookup(win, "F_SpinBoxXoffset", TRUE);
            widget_on(spin_box, EVT_VALUE_CHANGING, on_F_XoffsetChanging, win);
        }
        else if (strstr(name, "F_SliderXoffset"))
        {
            widget_t* slider = widget_lookup(win, "F_SliderXoffset", TRUE);
            widget_on(slider, EVT_VALUE_CHANGING, on_F_XoffsetChanging, win);
        }
. . .

xianjimli, 2019-10-14:
第一个问题我改了。是widget_set_value把value当unsigned处理了。

第二个问题，我这里正常： https://github.com/zlgopen/awtk-c-demos/blob/master/demos/spin_box.c

xianjimli, 2019-10-14:
你把spinbox的宽度调大点试试。

YinWenshuo, 2019-10-14:
@xianjimli 多谢回复，
第一个问题，更新代码后可以了。谢谢。
第二个问题，spin_box的宽度调很宽，现在是80，调到200都还是同样的显示问题。问题我再描述详细点：
按照我上面的xml，跑起来后，
2.1、如果先去拖动slider，那么spin_box的值显示就不正常，我拖动slider的时候，edit_set_int(spin_box,widget_get_value(target));//widget_get_value(target)得到slider的值，spin_box显示没有了个位数，但实际值是对的，(例如edit_set_int(spin_box,35)，显示3，但是此时再去操作spin_box的加或者减，spin_box就会正确地显示36或者34，说明刚开始拖动slider时，edit_set_int(spin_box,widget_get_value(target));是把值设进去了的，只是显示不对)
2.2、如果，先操作spin_box，再去拖动slider，则spin_box的值显示就正常，不会少了个位数的显示。

YinWenshuo, 2019-10-14:
还有个问题：
我的xml代码如下;
	<view >
		  <label x="0" y="middle" w="100" h="100%" text="radius"/>
		  <spin_box x="105" y="middle" w="80" h="100%" min="0.0" max="3.0" value="1.9" step="0.1" input_type="float" auto_fix="true" name="F_SpinBoxRadius" />
		  <slider x="190" y="middle" w="-210" h="100%" min="0.0" max="3.0" value="1.9" step="0.1" name="F_SliderRadius"/>
		</view>

widget_t* slider = widget_lookup(win, "F_SliderRadius", TRUE);
widget_set_value(slider,value);

==>当value设置为0.0~0.9时，slider的滑块直接滑到了最左端（即最小值处，类似于widget_set_value(slider,0);）
==>当value设置为1.0~1.9时，slider的滑块一直停留在widget_set_value(slider,1.0);处
==>当value设置为2.0~2.9时，slider的滑块一直停留在widget_set_value(slider,2.0);处
==>当value设置为3.0时，slider的滑块直接从widget_set_value(slider,2.0);处跳到widget_set_value(slider,3.0);处

（也就是说，slider在设置为小数时滑块还是整数的位置，不变）

YinWenshuo, 2019-10-14:
还有个问题：
我的xml代码如下;
<spin_box x="105" y="middle" w="80" h="100%" min="0.0" max="3.0" value="1.9" step="0.1" input_type="float" auto_fix="true" name="F_SpinBoxRadius" />
widget_t* slider = widget_lookup(win, "F_SliderRadius", TRUE);
widget_set_value(slider,value);
==>当value设置为0.0~0.9时，slider的滑块直接滑到了最左端（即最小值处，类似于widget_set_value(slider,0);）
==>当value设置为1.0~1.9时，slider的滑块一直停留在widget_set_value(slider,1.0);处
==>当value设置为2.0~2.9时，slider的滑块一直停留在widget_set_value(slider,2.0);处
==>当value设置为3.0时，slider的滑块直接从widget_set_value(slider,2.0);处跳到widget_set_value(slider,3.0);处
（也就是说，slider在设置为小数时滑块还是整数的位置不变，应该是slider的step没有改成float型，还是整型，所以滑块只能整数整数的跳，不能小数小数的跳）

xianjimli, 2019-10-14:
你更新代码试试。

YinWenshuo, 2019-10-14:
更新了代码，edit_set_int/set_double可以了。但是widget_get_value返回值仍然有问题，当slider的min、max取值是double时，widget_get_value（slider）返回的一直是0.0 。xml如下：

<view >
   <label x="0" y="middle" w="100" h="100%" text="radius"/>
   <spin_box x="105" y="middle" w="80" h="100%" min="0.0" max="3.0" value="1.9" step="0.1" input_type="float" auto_fix="true" name="F_SpinBoxRadius" />
   <slider x="190" y="middle" w="-210" h="100%" min="0.0" max="3.0" value="1.9" step="0.1" name="F_SliderRadius"/>
</view>

（好像没有看到slider_get_value接口函数）

xianjimli, 2019-10-14:
控件的属性标记为readable的，可以直接读取：https://github.com/zlgopen/awtk/blob/master/docs/manual/slider_t.md

YinWenshuo, 2019-10-14:
哦，通过widget_get_prop读取value。多谢！

closed 2019-10-14 by YinWenshuo