zlgopen/awtk #251: 请教几个问题？

ytchw101, 2019-10-14:
1.更新awtk最新源码（2019-10-14）（之前是201908-14）, 编译提示 在platform.c中
未定义clock_gettime(CLOCK_MONOTONIC, &ts);  和 usleep ，这两个在哪个文件定义的。

2.暂时注释掉1中未定义的接口，编译成功，下载代码到板子运行。打开有键盘的界面时，键盘不弹出来且程序直接卡死？ 不知什么原因？（我默认是进入有键盘的界面设置 mledit为focus 这样每次进入后键盘自动弹出 (老程序运行 一直可以的)）

3.早之前想使用freetype方式，后来windows编译时提示错误，现在看你们上次的更新后可以编译成功。 我是这样使用
(1)使用最新的awtk源码，.修改awtk_config.py 中 DWITH_STB_FONT  改为DWITH_FTFONT （不知是不是这样改？），编译源码。
(2).重新编译资源文件，然后替换工程中的资源文件。
(3)程序中定义宏：#define WITH_FT_FONT   1  ，编译工程。

xianjimli, 2019-10-14:
什么平台？

ytchw101, 2019-10-14:
嵌入式ARM

xianjimli, 2019-10-14:
Windows下，OpenGL模式不支持Freetype，需要使用AGGE模式。

xianjimli, 2019-10-14:
clock_gettime 和 usleep是linux标准函数啊

xianjimli, 2019-10-14:
```
VERSIONS
       These system calls first appeared in Linux 2.6.
```
你用的linux内核什么版本

ytchw101, 2019-10-14:
平台是ARM+RTT     我只是做应用    linux不懂...

xianjimli, 2019-10-14:
RTT不应该加入pc/platform.c

ytchw101, 2019-10-14:
哦  我再看看 我找同时来看看  谢谢

ytchw101, 2019-10-14:
另外 在这个平台使用freetype应该可以吧(之前一直用stb 是可以的)？  只是编译awtk源码 我修改awtk_config.py 中 DWITH_STB_FONT 改为DWITH_FT_FONT    可以吗？

xianjimli, 2019-10-14:
应该是可以的。

ytchw101, 2019-10-14:
使用freetype:
1.按照上面的修改awtk_config.py 中 DWITH_STB_FONT 改为DWITH_FT_FONT 编译源码完成后，然后重新生成资源， 程序中启用WITH_FT_FONT  编译后烧写，显示的字重叠在一起？

xianjimli, 2019-10-14:
我没在板子上用过freetype，明天我请同事帮忙看看。

soyzhc, 2019-10-18:
第三个问题我们这边验证，可以正常在windows和rt1052板子上运行，没有发现问题

closed 2019-11-04 by ytchw101