zlgopen/awtk #252: 关于字体解析和显示问题。

ytchw101, 2019-10-15:
遇到一个问题 就是界面创建一个label 然后widget_set_text(label, text);显示文字

当字号较大时有的字显示不出来？
比如字体大小都是192 解析汉字：
(1)汉字 “我” 显示不出来，但解析得到宽度*高度=152*149 也有数据，打印数据出来的确是 字模 数
    据 "我"。
(2) 汉字 “口” 可以显示，解析得到宽度*高度=124*125也有数据，都正确。

请问是什么问题？

我平台是：嵌入式ARM+RTT+LCD(480*272)

解析接口如下：
ret_t LoadChar(const wchar_t str, const char* font_name, uint8_t font_size, glyph_t* gy)
{
	 font_size_t size = GetFontSize(font_name);
	 font_t*  ft = font_manager_get_font(font_manager(), font_name, size)；
         return_value_if_fail(font_get_glyph(ft, str, font_size, gy) == RET_OK, RET_BAD_PARAMS);

         //打印
	 rt_kprintf("stb_get_glyph font_size=%d, w=%d,h=%d\r\n",font_size,g.w,g.h);
	int i, j;
	for(i = 0; i < g.h; i++)
	{	
		for(j = 0; j < g.w; j++)
		{
		   if(g.data[i*g.w + j])
			 {
			   rt_kprintf("@");
			 }
			 else
			 {
			   rt_kprintf("-");
			 }
		}
		
		rt_kprintf("\r\n");
	}
	

	 return RET_OK;
}

xianjimli, 2019-10-15:
控件多大，会不会比字体小

ytchw101, 2019-10-15:
我是先创建一个容器控件，容器控件高度是固定的 148，  然后label是基于容器创建的 ，label大小随着解析出来的字体宽高一致，  但是如果解析出来后  label高度超过了容器高度一点点 那么 label上的字 也不能显示吗？？   目前看好像是 我改到190 解析出来 150*148可以显示

ytchw101, 2019-10-15:
对于label的大小之前是这样设置的， 字体大时也是不显示
	widget_set_text(label, text); 
	label_resize_to_content(label,8,0xffff,8,0xffff); 

如果是一串汉字 “”  假如字号195有的字能显示有的显示不了，
所以是不是字型高度超过 label控件的父类(即容器控件)  也会导致不显示呢？

xianjimli, 2019-10-15:
不太清楚。你可以逐渐增加字体的大小，看多大就不正常了，或调试一下。

ytchw101, 2019-10-15:
我现在把容器控件和label都加大， 测试了几个字：

汉字    字体大小   解析宽度    解析高度    显示
“我 ”       190           150               148        正常
              191            151              149        不显示

“好  ”     191             148             147         正常
             192             149             149        不显示
 
“口”       214             138             139        正常
             215             139             140        不显示

还没搞懂不显示的字是什么原因。。。。。

xianjimli, 2019-10-15:
你更新一下代码：可能是glyph的x/y用的int8_t，字体太大导致溢出了，我改用int16_t了。

ytchw101, 2019-10-16:
可以了 感谢！

closed 2019-10-16 by ytchw101