zlgopen/awtk #254: 咨询，有1000个条目的list表初始化比较慢

poweel, 2019-10-16:
我们有一个电话联系人菜单，最大支持 2000 个条目，我们现在的做法是，在 xml 里面只留一个 list item，程序启动后 clone 2000 个这个 item，这种操作在 clone 的时候会比较慢，我们想优化下，能否给一些建议。

xianjimli, 2019-10-16:
这样试试：放在timer里，每个timer clone 100个，直到完成。

poweel, 2019-11-07:
我们现在测试clone 100个item要1s时间，clone 2000个要用20多s，放在timer里面clone也严重影响用户体验，感觉这个clone速度是有点异常，600M主频的cpu不应该这么慢。这个被clone的item是一个button里面放了一个image和2个label

xianjimli, 2019-11-07:
用的什么操作系统？
我在macos测试，每秒可创建40000个list item，每个list item里有一个图片，一个文本和一个开关。板子上即使慢100倍，每秒也可以创建400个啊。

这是我的测试程序： https://github.com/zlgopen/awtk-c-demos/blob/master/demos/list_item_clone.c

xianjimli, 2019-11-07:
如果板子上没有标准的malloc，那可能是内存分配器的问题，AWTK内置内存分配器是很简陋的，没有做小块内存的优化。

poweel, 2019-11-08:
linux系统，系统有malloc，在第一次创建后，如果使用WIDGET_POOL，后续创建的时候是不是就会很快了。
因为我们这个列表里面的控件属性都一样，是不是可以用类似memcpy的方法批量创建？

xianjimli, 2019-11-08:
WIDGET_POOL没有效果，后来去掉了。
profile一下，看看是哪里引起的。

xianjimli, 2020-01-01:
优化了一下，clone的时间降低为原来的1/3，你可以用新的代码试试。