zlgopen/awtk #261: 请教

ytchw101, 2019-10-23:
使用文件系统时，打开#define WITH_FS_RES  1宏 
每次在 tk_init里面的 return_value_if_fail(tk_init_internal() == RET_OK, RET_FAIL); 这句直接崩溃掉！ 不知什么原因？定位代码，运行到tk_init_internal里面的
return_value_if_fail(input_method_set(input_method_create()) == RET_OK, RET_FAIL); 这一句 崩溃 
查看之前备份的工程， （工程awtk源码是2019/8/14号之前的，不会崩溃 ）

soyzhc, 2019-10-23:
这个我们需要重现一下, 你使用干什么平台和系统, 
具体能不能跟进input_method_create函数看看是哪里崩溃

ytchw101, 2019-10-24:
搞定了！  是我把键盘UI资源等所有资源都放到SD卡里了， 导致input_method_create出问题 崩溃  
！！！！

xianjimli, 2019-10-24:
辛苦了，目前还没其他人这样用过，有空写个文档分享一下。

ytchw101, 2019-10-24:
好的 等过几天

ytchw101, 2019-10-24:
很好  很强大

ytchw101, 2019-10-24:
你好, 还有个问题请教下, try_load_assets函数是把所有资源预加载 到缓存中吧》？     有没有能够根据资源名称去释放该资源相关接口..        现在遇到个问题是
SD加的字体资源多的话，超过3个有的字体就不能使用，应该是缓存不够

xianjimli, 2019-10-24:
只加载用得的资源，目前字体不能卸载，可以卸载暂时不同的图片。

xianjimli, 2019-10-24:
2.如果我把同样的.ttf字体 改成不同的名字放在SD卡中, load_asset时是占两块缓存吗 还是同一块？

> 占用两块，没有必要去比较文件内容。 为什么要放两份？

ytchw101, 2019-10-25:
1.不能卸载？那么意思font_manager_unload_font卸载字体函数无用吗？ 我调用返回 0  应该成功了 可以吧？

2.两份是因为，只是做测试，没再去找一个.ttf字体。我测试是这样 本来是有3个不同字体在SD卡中， 程序3种字体使用显示都行，  然后又加了一个字体(和其中一个一样，只是名字不同)  使用这第四个字体始终无法使用， 应该是缓存不够，故想到卸载字体，卸载3个字体中和备份一样的那个字体后， 再使用备份字体  会出现崩溃了.....    卸载其他两个字体可以

xianjimli, 2019-10-25:
assets_manager里还有一份。要么就卸载全部字体，再重新加载。

ytchw101, 2019-10-29:
卸载步骤, font_xxx代表字体名称：
ret_t res = font_manager_unload_font(fm," font_xxx", 0);									
 asset_info_t* info = assets_manager_find_in_cache(rm, ASSET_TYPE_FONT,"font_xxx");
 res = assets_manager_unref(rm, info); 

使用上面 的步骤去卸载所有字体以及字体资源,然后再重新装载
1. 不知上面的卸载步骤对不对？
2.对于不是放在SD卡里的字体 ，卸载也是按照上面步骤吗？

 比如
#include "assets/inc/fonts/default.res"
....
assets_manager_add(rm, font_default); //默认加载一个字体防止 SD卡挂掉，无字体可使用

ytchw101, 2019-10-29:
对于上面的内置字体default.res，卸载了的话  怎么重新加载？

ytchw101, 2019-10-30:
清除assets_manager里面字体 是用assets_manager_unref 还是assets_manager_clear_cache？

closed 2019-10-31 by ytchw101