zlgopen/awtk #270: 关于矢量字体卸载、装载的几个疑问?

ytchw101, 2019-10-31:
总共4种字体， 其中msyh为内置字体,在通过在assets_init中添加到资源中如下：
#include "assets/inc/fonts/msyh.res" ...... 	assets_manager_add(rm, font_msyh);
并程序中指定为默认字体即	system_info_set_default_font(system_info(), "msyh");

另外3个字体放在SD卡中， 分别为 font1->2.7M左右，font2->2.2M左右， font3->2M左右

平台:嵌入式+RTT+LCD+SD卡  程序缓存设置约8M左右

现在是这样使用：默认上电只使用msyh字体，不加载SD卡中字体，只有在切换字体时，先卸载全部字体以及字体资源，然后再加载选中的字体，操作接口如下：

使用下面接口卸载全部字体，并加载一个字体作为使用：
void ReloadFont(const char* font_name)
{
		uint32_t i = 0;
		assets_manager_t* rm = assets_manager();
		font_manager_t* fm = font_manager();
		uint32_t nr = rm->assets.size;
		const asset_info_t** all = (const asset_info_t**)(rm->assets.elms);

                rt_kprintf("ReloadFont name:%s\r\n",font_name);
	
	         ret_t res = RET_FAIL;
		 for (i = 0; i < nr; i++) 
		 {
			const asset_info_t* iter = all[i];
			switch (iter->type) 
			{
			case ASSET_TYPE_FONT:
				if (iter->subtype == ASSET_TYPE_FONT_TTF) 
				{
				 res = font_manager_unload_font(fm, iter->name, 0);	//卸载字体	
				 rt_kprintf("---------------------------release font=%s, res:%d,",iter->name,res);
		                  asset_info_t* info = assets_manager_find_in_cache(rm, ASSET_TYPE_FONT, iter-
                                                                                                                  >name);
		                   ret_t ret = assets_manager_unref(rm, info);  //释放指定字体 iter->name资源
		                   rt_kprintf("ret=%d------\r\n",ret);
						
					}
				}
		 }	
		
		 assets_manager_clear_cache(rm, ASSET_TYPE_FONT); //清除字体类型缓存
		 if(strcmp(font_name, "msyh") != 0)
		 {
		      res= assets_manager_preload(assets_manager(), ASSET_TYPE_FONT, font_name);
		      rt_kprintf("assets_manager_preload name:%s, res=%d\r\n",font_name, res);
	         }
｝

几个问题请假：
1.当我卸载掉所有字体后，发现msyh字体还可以使用，是系统哪里自动加载默认字体吗？
2.对于上面的函数，没搞过不、也不知道是不是这个步骤，如有疑问，请指正一下？
3.现在字体间相互切换有个问题就是，相互间切换几次后，在加载字体font1(大小2.7M左右)时，即assets_manager_preload 这一句返回2（RT_FAIL失败），有时重新加载又成功，有时始终不行。不知道是何原因？ assets_manager_preload对加载的字体大小有要求吗？
4.使用过程中有时会导致程序崩溃卡死，从串口打印信息是：
6€〨S6€〨I6€〨G6€〨A6€〨B6€〨R6€〨T6€〨:6€〨 6€〨A6€〨b6€〨n6€〨o6€〨r6€〨m6€〨a6€〨
不知道啥原因？

xianjimli, 2019-10-31:
如果不支持中文输入法，可以只提取需要的字符的字体，字体文件就很小了，没有必要搞这么复杂。

ytchw101, 2019-10-31:
支持需要中文输入 ，         输入好  根据当前选择的字体    生成不同字体文字

xianjimli, 2019-10-31:
1.需要的时候会自动加载。
2.assets_manager_preload 失败可能时内存不够。

> 不要着急，静下心来，调试一下。

tangMans, 2019-10-31:
直接调用font_manager_load -> font_manager_add_font ->font_manager_unload_font看是否符合你的需求

ytchw101, 2019-10-31:
定位到问题     info = asset_info_create(type, subtype, name, size); 里面分配内存失败 ！的确是内存不够 

之前一直认为是不是没卸载完全导致！！  内存那一块包括文件系统都是同事搞的，以为留的空间足够。。。。。

ytchw101, 2019-10-31:
谢谢    我再琢磨下.....

closed 2019-11-04 by ytchw101