zlgopen/awtk #271: 控件文字的上下对齐问题

poweel, 2019-11-01:
我们使用了一些ttf内置的bdf字体（font_loader_ft引擎），结果出现有些控件显示正常，有些控件字体只能显示部分，控件高度为12，字体高度也为12，调试发现是设置的text_align_v不一样，按道理控件和字体高度一样，无论怎么对齐结果应该是一样的。调试发现这个字体的baseline=6，结果三种对齐对齐方式的结果就不一样了。这个用baseline的地方是不是应该用字体的最大高度？

canvas_draw_text_in_rect( )
  switch (c->text_align_v) {
    case ALIGN_V_TOP:
      y = r->y;
      break;
    case ALIGN_V_BOTTOM:
      y = r->y + (r->h - baseline);
      break;
    default:
      y = r->y + ((r->h - baseline) >> 1);
      break;
  }

xianjimli, 2019-11-03:
好的，谢谢，以前也没吃透，明天我研究一下。

poweel, 2019-11-03:
我们是用FontForge把bdf字库转换为ttf字库格式的，附件是原始的bdf字库和转换之后的ttf字库
[bdf2ttf.zip](https://github.com/zlgopen/awtk/files/3801503/bdf2ttf.zip)

xianjimli, 2019-11-03:
好的，谢谢

xianjimli, 2019-11-04:
完善了字体的处理，你有空试试。

closed 2019-12-12 by xianjimli