zlgopen/awtk #273: 关于输入法键盘字体对齐

ytchw101, 2019-11-04:
目前显示的键盘有的字母会超出对应的按钮区域， 所以想修改下文本对齐，但不知道在哪改？

1.试着修改ui资源中kb_default  ,比如字母q，   <button repeat="300" name="q" text="q" text_align_v="top"/> ，修改为向上对齐，不起作用。
2.  修改styles资源中keyboard,也不起作用
3.  键盘按钮风格和界面其他按钮风格应该是一致的，所以又修改styles资源中default, 修改为向上对齐， 普通按钮文字会向上对齐，键盘的按钮貌似更往下对齐！！？

xianjimli, 2019-11-04:
"有的字母会超出对应的按钮区域" 截图我看看。

ytchw101, 2019-11-04:
[image]

ytchw101, 2019-11-04:
上面文本设置风格和普通按钮一样居中对齐

在键盘高度固定情况下，感觉要让字模都显示在按钮区域：
1、要么修改每个按钮高度来适应文本。
2、要么修改文本对齐方式，如text_align_v 属性，

ytchw101, 2019-11-04:
刚才试了下text_align_v="bottom"时键盘按钮文字才上对齐 , 但上对齐后 大写字母又有好多超出按钮区域(在居中对齐时不会)

xianjimli, 2019-11-04:
我试了一下，可以生效啊：
demos/assets/default/raw/styles/keyboard.xml
```
  <style name="default" border_color="#a0a0a0"  text_color="black" text_align_v="top">
    <normal     bg_color="#f0f0f0" />
    <pressed    bg_color="#c0c0c0" x_offset="1" y_offset="1"/>
    <over       bg_color="#e0e0e0" />
    <focused    bg_color="#e0e0e0" />
    <disable    bg_color="gray" text_color="#d0d0d0" />
  </style>
```

xianjimli, 2019-11-04:
把字体改小一号。

closed 2019-12-12 by xianjimli