zlgopen/awtk #275: mono font和动画引起界面错乱

poweel, 2019-11-07:
我修改了lcd_mem_draw_glyph这个函数，来支持在windows和fb下应用mono字体，现在出现了一个问题是，如果主窗口里面有一个动画，再打开一个新窗口，然后在新窗口里面用鼠标随便扫过一个按钮（应该是触发重新渲染界面），在新窗口里和主窗口动画重叠的区域会出现字体渲染混乱。
调试发现是和，这个地方的sy，sx有关系，但是我没有看明白这个地方有什么问题
  uint8_t* src_p = glyph->data + glyph->w * sy + sx;
  if(glyph->format == GLYPH_FMT_MONO){
    src_p = glyph->data + glyph->pitch * sy + (sx >> 3);
  }  
我还实现了一个，如果字体颜色设置为全透明的话，就实现背景的反色显示字体，附件是我修改的patch和测试程序，还请帮忙看一下，谢谢！

[image]
[hello_awtk.zip](https://github.com/zlgopen/awtk/files/3817430/hello_awtk.zip)
[awtk.zip](https://github.com/zlgopen/awtk/files/3817433/awtk.zip)

xianjimli, 2019-11-07:
谢谢，明天我看看。

xianjimli, 2019-11-07:
谢谢，明天我看看。

xianjimli, 2019-11-08:
你为什么不按这个文档弄呢？ https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_mono_lcd.md

poweel, 2019-11-08:
我们这个方法很早就开始用了，那会还没有lcd_mono的支持，后面我试一下新的方法。

xianjimli, 2019-11-08:
好的。

closed 2019-12-03 by xianjimli