zlgopen/awtk #277: radio_button显示超长，能否让其滚动显示？

lizhenningi, 2019-11-14:
你好！
radio_button显示超长，之后我做了如下实验：
       使用hscroll_label作为其父控件让它来显示文字，并使用with_focus_state="true"让它能够在radio_button获得焦点时也获得焦点。
这样设置的结果：只有在刚进入此页面时hscroll_label会滚动，上下切换焦点之后，再次选中就不会滚动了。
出现这个结果的原因：当radio_button获得焦点时，父控件hscroll_label也随之获得焦点状态，但是父控件hscroll_label不会产生EVT_FOCUS事件。

xianjimli, 2019-11-15:
好的，谢谢。我想想怎么搞。

lizhenningi, 2019-11-15:
感谢您的回复！
在我使用AWTK的过程中，使用滚动显示的地方比较多，一般都是hscroll_label与其它控件结合使用，很少单独的用到hscroll_label，能不能给所有控件加上一个属性让其滚动显示。

xianjimli, 2019-11-15:
你更新一下代码。参考一下：https://github.com/zlgopen/awtk/blob/master/docs/how_to_scroll_you_text.md

lizhenningi, 2019-11-15:
好的，谢谢

lizhenningi, 2019-11-16:
您修改后的代码我试了，确实可以滚动显示，但是又出现了新的问题。
我的期望：我使用radio_button作为父控件，hscroll_label为其子控件。当radio_button有焦点时，hscroll_label的背景色由白色变为黑色、字体颜色由黑色变为白色。
问题：我发现当radio_button有焦点时，hscroll_label没有焦点状态（即：其style的状态没有改变），但是我又需要这个style的状态。

xianjimli, 2019-11-17:
好的，谢谢。我改了，根据父控件的焦点状态设置hscroll label的状态。你更新代码试试。

lizhenningi, 2019-11-19:
好的，可以用。谢谢！

closed 2019-11-19 by xianjimli