zlgopen/awtk #282: 焦点移动问题

cuimeng2213, 2019-11-15:
目前焦点移动只支持前后移动。 如何能修改为支持上下左右移动呢？

xianjimli, 2019-11-16:
我还在思考这个问题。

cuimeng2213, 2019-11-18:
我这有个思路， 可以根据移动方向计算可以聚焦的结点位置信息 返回最近的结点 然后设置该结点focused

xianjimli, 2019-11-18:
目前实现了上下移动垂直焦点，左右键切换水平焦点：https://github.com/zlgopen/awtk/tree/rework-focus

cuimeng2213, 2019-11-18:
好的 我更新代码试试。 你是这个开源代码的维护人员？

cuimeng2213, 2019-11-18:
不好意思你那个提交被我给点错关闭了。。

cuimeng2213, 2019-11-18:
功能已经实现

closed 2019-11-18 by cuimeng2213

xianjimli, 2019-11-18:
还测试一下，和软键盘结合起来，再合并到主分支。

cuimeng2213, 2019-11-18:
[image]
这样子写貌似会有问题，假如向右侧移动，右侧的焦点y轴与当前焦点位置y轴相差10 则不会移动焦点了。这个判断条件貌似需要修改一下。

xianjimli, 2019-11-18:
主要支持网格布局的控件。你的界面是什么样子的，贴张图我看看。

cuimeng2213, 2019-11-18:
[image]
布局是这样子的，每个都是一个button按钮

xianjimli, 2019-11-19:
你更新代码试试，rework-focus分支。

xianjimli, 2019-11-19:
如果不行，你把XML贴上来一下。

xianjimli, 2019-11-20:
改了，你参考：https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md