zlgopen/awtk #286: 数据生成图片显示请教

ytchw101, 2019-11-19:
1.根据自定义的图片数据pbmpdata 创建一个 bitmap_t 
bitmap_t* img=bitmap_create_ex(strMaxWidth,strMaxHeight,strMaxWidth*2,BITMAP_FMT_BGR565);
	img ->should_free_data = TRUE;
	img ->should_free_handle = TRUE;
	ret_t res = bitmap_init(img,w,h,BITMAP_FMT_BGR565,pbmpdata);

2.然后在EVT_PAINT中调用
      {
         TKMEM_ALLOC 重新申请pbmpdata  
          ........
	  canvas_draw_icon(c, img, w/2, h/2); 即可显示图片
          TKMEM_FREE(pbmpdata  )//释放 图片数据 
       }

想问下：我pbmpdata 是TKMEM_ALLOC 来动态申请  每次EVT_PAINT刷新图片显示后就释放掉
   但是貌似  TKMEM_FREE(pbmpdata  )没释放掉这个内存一样，  刷新一遍后 第二次TKMEM_ALLOC （同样数据大小）就申请不到？？？

bitmap_init后 对于graphic_buffer_t中的data应该和pbmpdata  指向同样内存吧？

ytchw101, 2019-11-19:
另外请问BITMAP_FMT_GRAY和BITMAP_FMT_MONO  之前问 是暂时不支持，后面有计划支持进去吗？

平台(嵌入式ARM+LCD)

xianjimli, 2019-11-20:
> 1.根据自定义的图片数据pbmpdata 创建一个 bitmap_t
> 想问下：我pbmpdata 是TKMEM_ALLOC 来动态申请 每次EVT_PAINT刷新图片显示后就释放掉
> 但是貌似 TKMEM_FREE(pbmpdata )没释放掉这个内存一样， 刷新一遍后 第二次TKMEM_ALLOC （同样数据大小）就申请不到？？？
> 
> bitmap_init后 对于graphic_buffer_t中的data应该和pbmpdata 指向同样内存吧？

控件销毁的时候再销毁吧，不要每次都重新创建。

ytchw101, 2019-11-20:
pbmpdata 不释放的话  我内存不够用  我不知道上面的用法有没什么问题？  实际使用上述方式绘制图片时 总觉得 哪地方有内存泄漏？  

图片删除时 调用
		  graphic_buffer_unlock(img->buffer);
		  graphic_buffer_destroy(img->buffer);
                  bitmap_destroy(img);
 稍微好些  但还是。。。  

还有什么要释放的？

ytchw101, 2019-11-20:
另外请问下 假如我图片数据data  大小1Mbyte 的话     那么  img->buffer  大概会占多少空间？

ytchw101, 2019-11-20:
这个由数据生成的图片 需要  image_manager_unload_bitmap 吗？

xianjimli, 2019-11-21:
> 另外请问下 假如我图片数据data 大小1Mbyte 的话 那么 img->buffer 大概会占多少空间？

img->buffer本身很小，看看代码就知道了。但是持续泄露又是另外一回事了。

xianjimli, 2019-11-21:
> 这个由数据生成的图片 需要 image_manager_unload_bitmap 吗？

不需要。

xianjimli, 2019-11-21:
这个分支对gray图片提供了有限的支持，你看对你有没有帮助： https://github.com/zlgopen/awtk/tree/gray_bitmap

closed 2019-11-22 by ytchw101