zlgopen/awtk #287: 焦点切换问题

zjm1060, 2019-11-21:
当窗口中同时含有button和edit两种控件的时候，用四方向键切换焦点，一旦焦点进入到edit后就再也切换不出来了

qianfan-Zhao, 2019-11-21:
[image]

我的也是使用方向键切换焦点。虚拟键盘关掉的时候，可以在次切换焦点。

但是在edit打开之后，界面上的button不显示了。输入的时候，字符也没有显示。可以看gif图片。

xianjimli, 2019-11-21:
> 当窗口中同时含有button和edit两种控件的时候，用四方向键切换焦点，一旦焦点进入到edit后就再也切换不出来了

改了，你更新代码试试。

xianjimli, 2019-11-21:
> [image]
> 
> 我的也是使用方向键切换焦点。虚拟键盘关掉的时候，可以在次切换焦点。
> 
> 但是在edit打开之后，界面上的button不显示了。输入的时候，字符也没有显示。可以看gif图片。
> 
> 我是把Return button从action改成close。close的时候，会关掉键盘。否则纯键盘操作，好像没法将焦点切换出去。
> 
> ```
>         <group_box children_layout="default(r=1,c=0,s=1,m=1)">
>           <button style="kb_ascii" focusable="true" name="page:symnum" w="30%" text="123"/>
>           <button style="kb_ascii" focusable="true" repeat="300" name="space" w="40%" text="Space"/>
>           <button style="kb_ascii" focusable="true" name="close" w="30%" text="Return"/>
>         </group_box>
> ```

我这里输入会显示，你把界面的XML贴上来我试试。

zjm1060, 2019-11-21:
焦点可以正常移动了，但如果控件布局不是上下左右对齐的话，焦点移动路劲比较怪异

qianfan-Zhao, 2019-11-21:
@zjm1060 焦点先行/列移动。如果不是同行，或者同列的话，选最近距离的。所以看上去有点诡异。

老哥在微信群不

zjm1060, 2019-11-21:
> @zjm1060 焦点先行/列移动。如果不是同行，或者同列的话，选最近距离的。所以看上去有点诡异。
> 
> 老哥在微信群不

倒也不是不能用，就是觉得不太符合人的预期

是 AWTK生态共建交流群，在里面

xianjimli, 2019-11-21:
> 左右对齐的话

你用最新的代码试试。

closed 2019-12-03 by xianjimli