zlgopen/awtk #288: 关于bitmap_create_ex控件以及输入法使用的两个疑问？

ytchw101, 2019-11-22:
1.使用bitmap_create_ex时即使最后删除了 貌似哪里还有内存泄漏？
 bitmap_t* img = NULL;
  for（i = 0; i < 10 ; i++）
 {
    printf("---------i=%d-------\r\n", i)；
   uint8_t* pdata =  TKMEM_ALLOC(2*1024*1024); //申请2M空间
   if(NULL == pdata)
   {
     printf("failed!\r\n");
     break;
   }
  else   //可以申请到
  {
      printf("success!\r\n");
      TKMEM_FREE(pdata); //释放掉
  }
  img = bitmap_create_ex(1024,128,128*2,BITMAP_FMT_BGR565);

  if(img)
  {
        graphic_buffer_unlockimg->buffer);
	graphic_buffer_destroy(img->buffer);
	rt_kprintf("graphic_buffer_destroy .....\r\n");
        bitmap_destroy(widget->img);
        img = NULL;
  }
 }

运行几次之后， 可能上面2M空间就申请不到了， 不知道为什么？

2.界面创建mledit之后，启动输入法键盘 ， 当我点击键盘上的关闭按钮后，再申请比如6M内存，可以申请成功， 但是当我使用 input_method_request(im, NULL)； 来关闭输入法， 再申请6M内存申请不到只能申请到大约5.2M  ， 当然当界面关闭或者说mledit被销毁后 同样是可以申请到6M的

所以想请问使用 input_method_request(im, NULL)后(之后添加
//  if (im->keyboard != NULL) {
//    window_manager_close_window_force(im->keyboard->parent, im->keyboard);
//    im->keyboard = NULL;
//  }
)也不行， 应该如何完全释放空间..... ?

xianjimli, 2019-11-22:
应该有内存碎片，没有连续6M的内存，就会失败

xianjimli, 2019-11-22:
外面不要调用graphic_buffer_destroy，可以debug一下bitmap_destroy看看。