zlgopen/awtk #289: STM32F407 移植后啥也不显示

673835452, 2019-11-22:
好像卡在demo_main.c 中的tk_run();位置
没移植输入

xianjimli, 2019-11-22:
看看stack size是多大。

673835452, 2019-11-22:
> 看看stack size是多大。
Stack_Size      EQU     0x00001000
Stack_Size      EQU     0x00002000

都试过了

平台原子407+4.3寸MCU屏

`typedef uint16_t pixel_t;

#define LCD_FORMAT BITMAP_FMT_BGR565
#define pixel_from_rgb(r, g, b)                                                \
  ((((r) >> 3) << 11) | (((g) >> 2) << 5) | ((b) >> 3))
#define pixel_to_rgba(p)                                                       \
  { (0xff & ((p >> 11) << 3)), (0xff & ((p >> 5) << 2)), (0xff & (p << 3)) }

#define set_window_func LCD_Set_Window
#define write_data_func LCD_WriteRAM
`
`

ret_t platform_prepare(void) {
  static bool_t inited = FALSE;
  static uint32_t s_heam_mem[1024*25];

  if (!inited) {
    inited = TRUE;
    tk_mem_init(s_heam_mem, sizeof(s_heam_mem));
  }

  return RET_OK;
}

`

`void TIM3_IRQHandler(void)
{
	if(TIM_GetITStatus(TIM3,TIM_IT_Update)==SET) //溢出中断
	{
		g_sys_tick++;
	}
	TIM_ClearITPendingBit(TIM3,TIM_IT_Update);  //清除中断标志位
}

uint64_t get_time_ms() {
  return g_sys_tick;
}

void sleep_ms(uint32_t ms) {
  uint32_t count = 0;
  uint32_t start = get_time_ms();

  while (get_time_ms() < (start + ms)) {
    count++;
  }
}`

`	u8 lcd_id[12];				//存放LCD ID字符串
	int a=0;
	NVIC_PriorityGroupConfig(NVIC_PriorityGroup_2);//设置系统中断优先级分组2
	delay_init(168);      //初始化延时函数
	uart_init(115200);		//初始化串口波特率为115200
	printf("UART_OK\r\n");
	LED_Init();					  //初始化LED
 	LCD_Init();           //初始化LCD FSMC接口
	
	POINT_COLOR=RED;      //画笔颜色：红色
	sprintf((char*)lcd_id,"LCD ID:%04X",lcddev.id);//将LCD ID打印到lcd_id数组。
	LCD_ShowString(30,110,200,16,16,lcd_id);	
	
	TIM3_Int_Init(2400,50);
	TIM4_Int_Init(2400,500);
	a = gui_app_start(480,800);//进入周立功的 AWTK
`

673835452, 2019-11-22:
运行 后 只显示LCD ID:5510

xianjimli, 2019-11-22:
支持PNG/jpg解码需要32K栈，改成0x8000试试。
参考一下：https://github.com/zlgopen/awtk/blob/master/docs/porting_common.md

673835452, 2019-11-22:
> 支持PNG/jpg解码需要32K栈，改成0x8000试试。
> 参考一下：https://github.com/zlgopen/awtk/blob/master/docs/porting_common.md

awtk_config.h 参考的103 
改成0x8000了 还是不行

xianjimli, 2019-11-24:
TIM3_IRQHandler里增加g_sys_tick应该不对的。要放到systick中断里。