zlgopen/awtk #291: 内存泄漏的问题

gaoyuan1111, 2019-11-23:
我们的代码中出现了大量的内存泄漏。通过memwatch调试。发现是awtk中出现了问题。出现的方法是一直通过按键进出一个界面，因为我们的程序控件过多运行不快，会出现在某个界面卡住一段时间的现象，这时会出现内存泄漏。猜测是因为，一直在open某个界面，但是因为上一个界面未关闭，阻塞了，显示不出来，但是却一直能创建控件导致的。请问有没有什么好的解决方案？下面是部分memwatch打印。
unfreed: <10123> awtk/src/tkc/mem.c(58), 8 bytes at 0x9ee9f4  	{64 65 66 61 75 6C 74 00 .. .. .. .. .. .. .. .. default.}
unfreed: <7946> awtk/src/tkc/mem.c(45), 8 bytes at 0x9d9b0c  	{00 00 00 00 94 9A 9D 00 .. .. .. .. .. .. .. .. ........}
unfreed: <7945> awtk/src/tkc/mem.c(45), 76 bytes at 0x9d9a94  	{00 00 00 00 01 00 00 00 01 00 00 00 00 00 00 00 ................}
unfreed: <3978> awtk/src/tkc/mem.c(45), 28 bytes at 0x995b04  	{10 00 00 00 05 00 00 00 06 00 00 00 88 B9 49 00 ..............I.}
unfreed: <3967> awtk/src/tkc/mem.c(58), 16 bytes at 0x9b42dc  	{09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................}
unfreed: <3965> awtk/src/tkc/mem.c(58), 16 bytes at 0x9b429c  	{09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................}
unfreed: <3963> awtk/src/tkc/mem.c(58), 40 bytes at 0x9b4244  	{44 00 00 00 69 00 00 00 73 00 00 00 61 00 00 00 D...i...s...a...}
unfreed: <3961> awtk/src/tkc/mem.c(58), 8 bytes at 0x9b420c  	{44 69 73 61 62 6C 65 00 .. .. .. .. .. .. .. .. Disable.}
unfreed: <3960> awtk/src/tkc/mem.c(45), 28 bytes at 0x9b41c4  	{FC 2D 9B 00 04 00 00 00 0E 01 00 00 D8 21 45 00 .-...........!E.}
unfreed: <3959> awtk/src/tkc/mem.c(45), 28 bytes at 0x9b417c  	{9C 57 99 00 01 00 00 00 0B 01 00 00 18 48 43 00 .W...........HC.}
unfreed: <3958> awtk/src/tkc/mem.c(45), 16 bytes at 0x9b413c  	{7C 41 9B 00 02 00 00 00 01 00 00 00 00 00 00 00 |A..............}
unfreed: <3957> awtk/src/tkc/mem.c(45), 28 bytes at 0x9b40f4  	{04 22 9B 00 01 00 00 00 0B 01 00 00 74 2A 45 00 ."..........t*E.}
unfreed: <3956> awtk/src/tkc/mem.c(45), 16 bytes at 0x9b40b4  	{F4 40 9B 00 02 00 00 00 01 00 00 00 00 00 00 00 .@..............}

xianjimli, 2019-11-23:
mem.c只是包装了系统的malloc函数，把callstack打出来，才知道是谁真正调用的malloc函数。

最好的办法就是用awtk-linux-fb版本，在valgrind下测试，内存泄漏无所遁形。

gaoyuan1111, 2019-11-26:
我找到内存泄漏的问题是因为filename过长导致的。但是我修改了长度后，open_window的时候会出现断错误，但是我找到了断错误发生的位置却不知道什么导致的。请问您知道是什么原因造成的吗？/*TK_NAME_LEN+1 must aligned to 4*/enum { TK_NAME_LEN = 31 };。。31改为63。
[image]

xianjimli, 2019-11-26:
用valgrind跑一下。

gaoyuan1111, 2019-11-26:
[memleak.txt](https://github.com/zlgopen/awtk/files/3890328/memleak.txt)
这个是调试出来的问题。加了FT_Done_Glyph(glyph);这个以后 还是不顶用

xianjimli, 2019-11-26:
加个按钮，点击时调用tk_quit正常退出应用程序，强行关闭有些泄露是正常现象。

closed 2020-05-25 by gaoyuan1111