zlgopen/awtk #293: 当每个界面都有一个butotn的x、y、w、h是一样的，能否通过设置头文件的方式将这个button包含进去，需要调用的界面只需调用头文件里定义的名字，这样我想改它的x、y、w、h时，只需改头文件就好了

zhaoxin-1017, 2019-11-25:


xianjimli, 2019-11-25:
可以在xml中include。

xianjimli, 2019-11-25:
```
auto_play.xml:    <?include filename="slide_view.inc" ?>
auto_play_landscape.xml:    <?include filename="slide_view.inc" ?>
auto_play_portrait.xml:    <?include filename="slide_view.inc" ?>
menu_down_center.xml:  <?include filename="menu.inc" ?>
menu_down_left.xml:  <?include filename="menu.inc" ?>
menu_down_right.xml:  <?include filename="menu.inc" ?>
menu_left_bottom.xml:  <?include filename="menu.inc" ?>
menu_left_middle.xml:  <?include filename="menu.inc" ?>
menu_left_top.xml:  <?include filename="menu.inc" ?>
```

zhaoxin-1017, 2019-11-25:
这个inc文件放在ui目录下吗？还是放在inc目录下的data或者xml里？还是说在raw目录下重建一个inc目录？定义的格式可以这样吗<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  style="lv2_menu_left_define_" focusable="true" focus="true"/>，引用为<button name="wifi_settings_wifi"/>

xianjimli, 2019-11-25:
在ui目录下，参考一下demos。

zhaoxin-1017, 2019-11-25:
请问能不能通过修改c代码给控件设置一个属性widget_style，需要调用头文件里的控件时通过这个属性找到我所要调的控件？auto_play.xml:    <?include filename="slide_view.inc" widget_style=“aaa” ?>

xianjimli, 2019-11-25:
可以设置一个自定义的属性，然后用foreach遍历找到这类控件。

zhaoxin-1017, 2019-11-25:
就是说在属性的设置里添加foreach遍历，以便调用的地方找到；还是说只是设置了一个属性，在调用的地方foreach遍历找到？

zhaoxin-1017, 2019-11-25:
这样：main.inc
<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  widget_style="lv2_menu_left_define_1" focusable="true" focus="true"/>
<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  widget_style="lv2_menu_left_define_2" focusable="true" focus="true"/>
<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  widget_style="lv2_menu_left_define_3" focusable="true" focus="true"/>
调用处：
<?include filename="main.inc" widget_style="lv2_menu_left_define_1" ?>是可以实现的对吗

zhaoxin-1017, 2019-11-25:
这样：main.inc
`<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  widget_style="lv2_menu_left_define_1" focusable="true" focus="true"/>
<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  widget_style="lv2_menu_left_define_2" focusable="true" focus="true"/>
<button name="wifi_settings_wifi"  tr_text='1.Wi-Fi'	  widget_style="lv2_menu_left_define_3" focusable="true" focus="true"/>
`
调用处：
`<?include filename="main.inc" widget_style="lv2_menu_left_define_3" ?>`
是可以通过设置widget_style实现的对吧

xianjimli, 2019-11-26:
<?include filename="main.inc" ?>只是把main.inc原封不动的展开到这里，widget_style="lv2_menu_left_define_3" 这部分在这没作用(或者导致语法错误)。

zhaoxin-1017, 2019-11-26:
请问通过修改.c将x、y、w、h这四个选项加到style的属性设置中可以吗？类似这样
`<style name="show_time" text_color="white" x="c" y="10" w="25" h="25">
        <normal bg_image="time"/>
    </style> `

xianjimli, 2019-11-26:
目前不支持，我考虑一下。

evilbinary, 2019-12-04:
简单来说支持公用样式嘛

zhaoxin-1017, 2019-12-10:
最新代码有相关的实现吗？

evilbinary, 2019-12-10:
@zhaoxin-1017 目前没有，李老师说，工作量很大。

xianjimli, 2019-12-11:
支持了，请参考这个文档： https://github.com/zlgopen/awtk/blob/master/docs/how_to_write_layout_params_in_style.md

zhaoxin-1017, 2019-12-11:
好的，谢谢

evilbinary, 2019-12-11:
@xianjimli 可以啊，不过x y 是不是也可以用style 控制？

xianjimli, 2019-12-11:
> 不过x y 是不是也可以用style 控制？

可以啊。

evilbinary, 2019-12-11:
好，李老师辛苦了