zlgopen/awtk #307: 使用combo_box为父控件，hscroll_label为子控件，进程会死掉

lizhenningi, 2019-12-11:
您好！
这是对应的xml代码：
`<window theme="main" move_focus_prev_key="up" move_focus_next_key="down">
	<button name="button3" x="0" y="90" w="100%" h="30" text="33333333333333333333" focusable="true" style="default"/>
	<button name="button4" x="0" y="120" w="100%" h="30" text="44444444444444444444" focusable="true" style="default"/>
	<button name="button1" x="0" y="150"  w="100%" h="30" style="default" with_focus_state="true">
		<combo_box name="unconditional" x="170" y="2" w="140" h="30" options="Disable;Enable;" focusable="true" readonly="true">
			<hscroll_label  x="-160" y="-1"  w="140" h="30" tr_text="7777777777777777777777" loop="true" duration="10000" lull="1000" only_parent_focus="true"/>
		</combo_box>
	</button>
</window>`
将父控件换成button，则进程不会死，且进程是在调用window_open函数打开此页面时死掉的

xianjimli, 2019-12-11:
你想实现什么样的效果

lizhenningi, 2019-12-11:
想要实现的效果：
当combo_box有焦点时hscroll_label滚动显示。
当combo_box失去焦点时，hscroll_label停止滚动并回到初始显示状态。

xianjimli, 2019-12-11:
我改了一下，你参考这个文档试试： https://github.com/zlgopen/awtk/blob/master/docs/manual/combo_box_t.md
````
  <combo_box left_margin="6" readonly="true" x="10" y="50" w="80" h="30" options="leftttttttttt;centerrrrrrrrrrrrrrrr;rightttttttttt;"
  selected_index="1">
  <hscroll_label x="0" y="0" w="-30" h="100%"
    name="value"
    lull="1000"
    loop="true"
    yoyo="true"
    ellipses="true"
    only_parent_focus="true"/> 
  <button style="combobox_down" x="right:5" y="middle" w="20" h="20"/>
</combo_box>
```

lizhenningi, 2019-12-11:
好的，谢谢

lizhenningi, 2019-12-11:
您好！
您的修改生效了，但是combo_box中的图标显示不出来了

xianjimli, 2019-12-12:
我改了，你更新代码试试。谢谢

lizhenningi, 2019-12-12:
您好！
修改生效了，但是出现了控件style不设置背景颜色或图片的话默认显示为黑色的情况，与修改前的显示不相同

xianjimli, 2019-12-12:
应该是不同的，hscroll label挡住了combo box。你要给hscroll label设置style，去掉hscroll label的背景色试试。

lizhenningi, 2019-12-12:
您好！
现在不只是hscroll label与combo box的问题，其它控件也是不设置背景颜色或图片的话默认显示为黑色。
在本次修改之前的逻辑是：子控件不设置背景颜色或图片其背景是透明的，显示的是父控件的背景。
在本次修改之后：子控件不设置背景颜色或图片其背景是黑色的，将父控件的背景给覆盖掉了。
现在我们的项目是基于本次修改之前的逻辑写的，能否还是用本次修改之前的显示逻辑。

lizhenningi, 2019-12-12:
您好！
上一条提问所造成的问题原因，是我操作不当造成的。如果给您造成困扰，我非常抱歉

closed 2020-01-01 by xianjimli