zlgopen/awtk #309: windows和linux环境搭建问题

Jone-Zheng, 2019-12-16:
windows环境搭建问题：
Python38 + scons-3.1.1 + Visual Studio 2015 
编译awtk-v1.3+rc.1@aworks.2，出现错误：
[image]

linux环境搭建：
按手册说明搭建环境，编译环境是i.mx6 yocto工程拉去的：fsl-imx-x11-glibc-x86_64-meta-toolchain-qt5-cortexa9hf-neon-toolchain-4.9.11-1.0.0.sh

编译awtk-v1.3+rc.1，出现错误：
[image]

还请高人赐教。

Jone-Zheng, 2019-12-16:
由于图标有问题，补充图片内容：
windows下的错误：
MACH=AMD64 ARCH=('64bit', 'WindowsPE') TARGET_ARCH=
TK_ROOT: F:\awtk\awtk\awtk
scons: done reading SConscript files.
scons: Building targets ...
cl /Fo3rd\agge\agge\math.obj /c 3rd\agge\agge\math.cpp /TP -DWIN32 -D_WIN32 -DWI
NDOWS /EHsc -D_CONSOLE /DEBUG /Od /FS /Z7 -DWITH_64BIT_CPU -DTK_ROOT=\"F:\\awtk\
\awtk\\awtk\" -DSTBTT_STATIC -DSTB_IMAGE_STATIC -DWITH_STB_IMAGE -DWITH_SOCKET -
DWITH_VGCANVAS -DWITH_UNICODE_BREAK -DWITH_DESKTOP_STYLE -DSDL2 -DHAS_STD_MALLOC
 -DWITH_SDL -DWITH_FS_RES -DHAS_STDIO -DHAVE_STDIO_H -DWITH_STB_FONT -DWITH_BITM
AP_BGRA -DWITH_NANOVG_SOFT -DWITH_FB_BGR565=1 -DWITH_NANOVG_AGGE -D_WIN64 -DSDL_
REAL_API -DSDL_HAPTIC_DISABLED -DSDL_SENSOR_DISABLED -DSDL_JOYSTICK_DISABLED -D_
_STDC_LIMIT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_CONSTANT_MACROS -D_HAS_EXCEPT
IONS=0 -D_HAS_ITERATOR_DEBUGGING=0 -D_ITERATOR_DEBUG_LEVEL=0 -D_SCL_SECURE=0-D_S
ECURE_SCL=0 -D_SCL_SECURE_NO_WARNINGS -D_CRT_SECURE_NO_WARNINGS -D_CRT_SECURE_NO
_DEPRECATE /I. /Isrc /I3rd /Isrc\ext_widgets /I3rd\pixman /I3rd\cairo /I3rd\bgfx
\bgfx\include /I3rd\bgfx\bx\include /I3rd\bgfx\bimg\include /I3rd\agge /I3rd\agg
\include /I3rd\nanovg /I3rd\nanovg\gl /I3rd\nanovg\base /I3rd\nanovg\agge /I3rd\
nanovg\bgfx /I3rd\SDL\src /I3rd\SDL\include /I3rd\agge\src /I3rd\agge\include /I
3rd\gpinyin\include /I3rd\libunibreak /I3rd\gtest\googletest /I3rd\gtest\googlet
est\include /Itools
scons: *** [3rd\agge\agge\math.obj] Error 309
scons: building terminated because of errors.

linux下的错误：
zy@zy-mac:~/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk-linux-fb$ ls ../user_apps/HelloWorld-Demo/
assets_c_gen.py  assets_gen.bat  assets_gen.py  assets_gen.sh  bin  res  SConstruct  src  web.json
zy@zy-mac:~/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk-linux-fb$ scons APP=../user_apps/Helloworld-Demo
scons: Reading SConscript files ...

scons: warning: Ignoring missing SConscript 'build/var/Helloworld-Demo/src/SConscript'
File "/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk-linux-fb/SConstruct", line 45, in <module>
scons: done reading SConscript files.
scons: Building targets ...
/opt/fsl-imx-x11/4.9.11-1.0.0/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc -o awtk-port/input_dispatcher.o -c -std=gnu99 -g -Wall -Os -mfloat-abi=hard -DHAS_STD_MALLOC -DWITH_FS_RES -DHAS_STDIO -DWITH_VGCANVAS -DWITH_UNICODE_BREAK -DLINUX -DSTBTT_STATIC -DSTB_IMAGE_STATIC -DWITH_STB_IMAGE -DWITH_STB_FONT -DWITH_NANOVG_AGGE -DWITH_WIDGET_TYPE_CHECK -DLINUX -DHAS_PTHREAD -DENABLE_CURSOR -DHAS_TSLIB -I/opt/tslib1.4/include -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src/ext_widgets -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/tools -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd/agge -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd/agg/include -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd/nanovg -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd/nanovg/base -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd/libunibreak -I/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/3rd/gpinyin/include awtk-port/input_dispatcher.c
In file included from /home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src/tkc/slist.h:25:0,
                 from /home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src/tkc/idle_manager.h:25,
                 from /home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src/base/idle.h:25,
                 from /home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src/base/event_queue.h:25,
                 from awtk-port/input_dispatcher.h:25,
                 from awtk-port/input_dispatcher.c:22:
/home/zy/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk/src/tkc/types_def.h:25:18: fatal error: math.h: No such file or directory
 #include <math.h>
                  ^
compilation terminated.
scons: *** [awtk-port/input_dispatcher.o] Error 1
scons: building terminated because of errors.
zy@zy-mac:~/fsl-release-bsp/awtk/awtk-v1.3+rc.1/awtk-linux-fb$

soyzhc, 2019-12-16:
感觉是编译环境没有安装完整的问题

可以把你的开发环境信息打印出来，发上来看看

- 在cmd命令行，运行 python
- 在cmd命令行，运行 scons
- 在VS2015开发人员命令行，运行 cl

或者下载下面的awtk-tools-checker工具，在windows下检查看看
```
https://pan.baidu.com/s/1synMvLTXXT3RZDjMH9a-pw
```

Jone-Zheng, 2019-12-17:
> 感觉是编译环境没有安装完整的问题
> 
> 可以把你的开发环境信息打印出来，发上来看看
> 
> * 在cmd命令行，运行 python
> * 在cmd命令行，运行 scons
> * 在VS2015开发人员命令行，运行 cl
> 
> 或者下载下面的awtk-tools-checker工具，在windows下检查看看
> 
> ```
> https://pan.baidu.com/s/1synMvLTXXT3RZDjMH9a-pw
> ```

已解决，非常感谢！
windows下占时未找到编译出错的原因，但是换了台电脑，按顺序安装，可以正常编译。
linux下的环境，需要修改awtk_config.py文件，具体修改如下：
[image]
[image]
因为全局变量已经定义了CC CXX LD AR STRIP，所以这个配置文件里可以去掉这些变量的定义。
[image]

xianjimli, 2019-12-17:
好的，谢谢分享

closed 2020-01-01 by xianjimli

soyzhc, 2020-03-15:
windows上编译 error 309 的问题以及重现出来
主要是因为新的win10上安装的 vc_redist 运行时库有问题, 导致无法执行cl编译程序
在 powershell上运行scons只有 error 309提示, 但可以在 cmd.exe 上运行可以获得更多错误信息

解决方案二选一:  
1 在控制面板"卸载程序" 里面, 修复 Visual C++ 2015 Redistributable
2 重新安装  vc_redist.x64.exe, vc_redist.x86.exe