zlgopen/awtk #311: 关于画布的问题

seventiger, 2019-12-17:
有个功能需要通过在画布上绘制文字，然后获取画布数据并进行二值化，目前遇到了2个问题：
1.LCD的分辨是320X240的，想创建一个比这个分辨率大的画布，比如640*480的，请帮忙指导下该怎么创建？
2.要求整个画布及绘制内容不显示，只在后台完成就行了，这个该如何实现？
  谢谢！

xianjimli, 2019-12-18:
可以的。

```
* 1.创建lcd对象。参考src/lcd/目录下的文件，选择需要的格式。
* 2.创建canvas对象。参考src/base/canvas.h
* 3.把lcd对象中的buffer取出来，创建一个bitmap对象。
```

你先试试，如果有问题，及时反馈

seventiger, 2019-12-20:
我现在的做法是：
   仿照真实的LCD参数创建了一个虚拟的LCD，然后创建canvas，画布使用canvas_begin_frame(c, &r, LCD_DRAW_OFFLINE);绘制结束后
       lcd_mem_t* mem = (lcd_mem_t*)lcd;
       uint8_t* fbuff = mem->offline_fb;
取fbuff 里面的数据（RGBA8888），发现数据是错误的。
请问我的流程是不是有问题？

xianjimli, 2019-12-20:
把完整代码贴上来看看

seventiger, 2019-12-20:
lcd_t* lcd_vritual_create(void)
{
    lcd_t* lcd = NULL;
    fb_info_t* fb = &v_fb;
    uint32_t size = 0;

    memset(fb, 0x00, sizeof(fb_info_t));
    fb->var.xoffset = 0;
    fb->var.yoffset = 0;
    fb->var.xres = 384;
    fb->var.yres = 480;
    fb->var.xres_virtual = fb->var.xres * 1;
    fb->var.yres_virtual = fb->var.yres * 1;
    fb->var.bits_per_pixel = 32;
    fb->fbmem0 = (uint8_t*)malloc(size);
    fb->var.red.offset = 0;
    fb->var.red.length = 8; 
    fb->var.green.offset = 8;
    fb->var.green.length = 8;
    fb->var.blue.offset = 16;
    fb->var.blue.length = 8;
    
    fb->fix.xpanstep = 0;
    fb->fix.ywrapstep = 0;
    fb->fix.line_length = fb->var.xres * fb->var.bits_per_pixel;
    fb->fix.smem_len = fb_size(fb)*3;
    lcd = lcd_linux_create(fb);
    return lcd;
}

static void canvas_vritual_init(const char* name, widget_t* to_close) {
  int i = 0,j = 0;
  canvas_t c;
  lcd_t* lcd = lcd_vritual_create();
  canvas_init(&c, lcd, font_manager());
  canvas_paint_text(&c);
}

static ret_t canvas_paint_text(canvas_t *c)
{
    uint32_t i = 0;
    vgcanvas_t* vg;
    rect_t r = {0,0,384,480};
    color_t txtColor = color_init(0, 0, 0, 0xff);
    color_t bkgColor = color_init(0xff, 0xff, 0xff, 0xff);

    canvas_begin_frame(c, &r, LCD_DRAW_OFFLINE);
    vgcanvas_save(vg);
    
    vgcanvas_translate(vg, 0, 0);
    canvas_set_fill_color(c, bkgColor);
    canvas_fill_rect(c, 0, 0, 384, 480);
    
    vgcanvas_translate(vg, 20, 20);
    canvas_set_text_color(c, txtColor);
    canvas_set_font(c, NULL, 30);
    canvas_draw_text(c, L"123456", 6, 0, 0);
    
    vgcanvas_restore(vg);

    canvas_end_frame(c);

    return RET_OK;
}

xianjimli, 2019-12-20:
不用这么麻烦，用lcd_mem_bgr565_create_single_fb创建就好。