zlgopen/awtk #313: edit框的输入提示的字体颜色

huamiao1996, 2019-12-23:
当设置tips属性的时候，这个输入的字符字体颜色通过text_color属性设置。但是设置的输入提示tips属性的字体颜色支持设置吗

xianjimli, 2019-12-23:
用这两个状态下的文本颜色：

```xml
    <empty      bg_color="#f0f0f0" text_color="#a0a0a0" />
    <empty_focus bg_color="#f0f0f0" text_color="#a0a0a0" border_color="black"/>
```

huamiao1996, 2019-12-23:
这个edit框内的光标也是灰色的，支持光标的颜色和输入字体的颜色是一致的吗

用这两个状态下的文本颜色：

xianjimli, 2019-12-23:
光标的颜色和输入字体的颜色是一致的

huamiao1996, 2019-12-24:
光标的颜色没有和输入字体的颜色一致，而是和tips的颜色一致，是我参数设置的有问题吗？

xianjimli, 2019-12-24:
是我没说清楚。输入字体和tips都是edit显示的文本，只是在不同状态下显示，所以它们的颜色都是由text_color决定。光标的颜色和当前状态下的文本颜色一致的。

huamiao1996, 2020-02-07:
你好，打扰了。您2020.1.1号的这两次关于克隆函数的修改是减少了克隆所需要的时间吗

| |
huamiao186
|
|
邮箱：huamiao186@163.com
|

Signature is customized by Netease Mail Master

在2019年12月24日 10:04，jim 写道：

是我没说清楚。输入字体和tips都是edit显示的文本，只是在不同状态下显示，所以它们的颜色都是由text_color决定。光标的颜色和当前状态下的文本颜色一致的。

xianjimli, 2020-02-07:
是的

huamiao1996, 2020-03-13:
您好，麻烦问一个问题。awtk-port目录下会创建一个tslib_thread的线程来获取触摸数据核心。这个线程调用标准的tslib，这个tslib都有什么功能呢？

| |
huamiao186
|
|
邮箱：huamiao186@163.com
|

Signature is customized by Netease Mail Master

在2020年02月07日 18:37，jim 写道：

是的