zlgopen/awtk #314: xml文件中 设置<?xml version="1.0" encoding="gb2312"?>仍然无法显示中文

chensheng-GitHub-2019, 2019-12-23:
项目中多处文本需要用到中文，测试xml
<?xml version="1.0" encoding="gb2312"?>
<window style="main" x="0" y="0" w="692" h="570" windowTitle="MainWindow">
<view name="centralwidget" x="0" y="0" w="100%" h="100%">
<label name="label" x="270" y="50" w="291" h="61" layoutDirection="Qt::LeftToRight" text="请输入用户名个密码" alignment="Qt::AlignCenter" />
<button name="pushButton" x="250" y="170" w="321" h="91" text="确认" />
<button name="pushButton_2" x="260" y="300" w="341" h="71" text="cancel" />
<combo_box name="comboBox" x="290" y="420" w="281" h="81" readonly="true" tr_text="user" options="1:user;2:user123用户;3:用户;4:123"/>
</view>
</window>

label 无法显示，”确认”，显示“确”  user123用户，显示“user123” 第三项空白
awtk 的xml文件有办法显示中文吗？

chensheng-GitHub-2019, 2019-12-23:
测试xml被吃了，再贴一次
<?xml version="1.0" encoding="gb2312"?>
<window style="main" x="0" y="0" w="692" h="570" windowTitle="MainWindow">
<view name="centralwidget" x="0" y="0" w="100%" h="100%">
<label name="label" x="270" y="50" w="291" h="61" layoutDirection="Qt::LeftToRight" text="请输入用户名个密码" alignment="Qt::AlignCenter" />
<button name="pushButton" x="250" y="170" w="321" h="91" text="确认" />
<button name="pushButton_2" x="260" y="300" w="341" h="71" text="cancel" />
<combo_box name="comboBox" x="290" y="420" w="281" h="81" readonly="true" tr_text="user" options="1:user;2:user123用户;3:用户;4:123"/>
</view>
</window>

chensheng-GitHub-2019, 2019-12-23:
[image]

xianjimli, 2019-12-23:
目前只支持UTF-8编码