zlgopen/awtk #316: 控件中直接写样式属性设置无效

webnoise, 2019-12-25:
如awtk/demos/assets/default/raw/ui/edit.xml中的edit控件right_margin属性实际无效，导致使用起来很不方便。

<edit w="70%" right_margin="16" tips="uint(0-150) auto_fix" input_type="uint" min="0" max="150" step="1" auto_fix="true" style="number" focused="true">
        <button name="inc" repeat="300" style="spinbox_up" x="right" y="0" w="15" h="50%" />
        <button name="dec" repeat="300" style="spinbox_down" x="right" y="bottom" w="15" h="50%" />
      </edit>

webnoise, 2019-12-25:
[
[image]
](url)

webnoise, 2019-12-25:
[image]

xianjimli, 2019-12-25:
我这里没问题啊。把你的XML贴上来看看。

webnoise, 2019-12-25:
一行代码都没改，awtk/demos/assets/default/raw/ui/edit.xml原生的，在mac系统上也有这个问题。

xianjimli, 2019-12-25:
那可能是我们对right_margin的理解不一样。right_margin在这里的目的是留一部分空间给按钮用，不会影响边框的大小。

你希望什么样的效果？

webnoise, 2019-12-25:
对于这个问题，期望的效果是控件的右边留出一些白边。
如上仅仅是个原生代码的一个例子。

对于如下代码，配置文本颜色或文字大小也是无效的，放在style里面就可以。
<label x="0" y="0" w="100%" h="10%" text="up/down key change focus1" text_color="red" font_size="72"/>

xianjimli, 2019-12-25:
* inline style请参考这个文档：https://github.com/zlgopen/awtk/blob/master/docs/theme.md

```
style:text_color="green"
```

* 『留出一些白边』，修改宽度即可。

webnoise, 2019-12-25:
个人觉得需要加个　style:　有些多余了，近期做xml布局开发，有几个建议。
1. 建议增加xml运行时错误校验。
2. 文本宽度、高度，建议增加类似安卓的wrap_content布局。
3. 建议兼容安卓应用xml布局，减少学习成本，提高布局灵活性。

closed 2019-12-25 by webnoise