zlgopen/awtk #317: 关于输入法——候选字控件的问题

lizhenningi, 2019-12-26:
您好！
能否让候选字控件能够通过方向键与数字键切换选择需要的候选字？

xianjimli, 2019-12-26:
好的，谢谢，我考虑一下。

xianjimli, 2019-12-30:
请参考这个文档修改: https://github.com/zlgopen/awtk/blob/master/docs/manual/candidates_t.md

lizhenningi, 2019-12-31:
您好！
我参考candidates_t.md做了修改，左右键切换选择可以使用，但是不能使用数字键选择候选字

xianjimli, 2020-01-01:
> 您好！
> 我参考candidates_t.md做了修改，左右键切换选择可以使用，但是不能使用数字键选择候选字

你指软件盘上的数字键吗？我用键盘是可以的啊。

lizhenningi, 2020-01-02:
您好！
不好意思，是我对问题没有描述准确！
问题是这样的：在使用中文输入法输入拼音后，候选框显示备选字，这时可以通过数字键选择备选框中的候选字。
在参考candidates_t.md做了修改后，使用中文输入法输入拼音，候选框显示备选字，之后在软键盘上点击数字键切换按钮“123”，之后出现数字键，点击备选字对应的数字键，edit控件中显示的是数字而不是对应的备选字

xianjimli, 2020-01-02:
你更新一下代码，数字键盘这样写：

```xml
        <button repeat="300" name="hard_key:1" text="1"/>
        <button repeat="300" name="hard_key:2" text="2"/>
        <button repeat="300" name="hard_key:3" text="3"/>
````

参考： https://github.com/zlgopen/awtk/blob/master/docs/manual/keyboard_t.md

lizhenningi, 2020-01-02:
可以了，谢谢您

closed 2020-05-21 by xianjimli