zlgopen/awtk #318: 关于idle_queue使用问题？

ytchw101, 2019-12-28:
请问下线程中频繁使用idle_queue来操作控件，  使用完后有什么要注意的地方吗？比如释放什么或者需要销毁啥吗？   感觉循环10000次后， 界面操作变的有些卡顿，不知何原因？   当然也不一定是使用idle_queue导致的问题，  还在排查.....

xianjimli, 2019-12-29:
用这个demo测试一下：https://github.com/zlgopen/awtk/blob/master/demos/demo_thread_app.c

YinWenshuo, 2020-09-23:
我也有同样的疑问，idle_queue后需要destory吗，在idle_queue里面的回调函数需要返回RET_REMOVE吗？

void* tk_thread_entry(void* args)
{
    while(1)
    {
        idle_queue(on_reflash_UI_idle,args);
        sleep_ms(5);
    }

    return NULL;
}

1、on_reflash_UI_idle这个回调函数里面返回RET_OK?
2、这里如果不sleep_ms（5）的话，ui这个进程占用cpu会很高，但如果sleep_ms(5)的话，会存在有情况没有被处理的情况。

ytchw101, 2020-09-23:
1.是返回RET_OK （目前我程序线程中 这样使用   循环1万次左右 没出现啥问题  ）idle_queue后不需要destory2.可以搞个信号量  什么的啊

xianjimli, 2020-09-23:
如果希望重复执行就返回RET_REPEAT，否则返回RET_OK或RET_REMOVE。如果希望重复执行，返回RET_REPEAT就行了，idle_queue只需调用一次，都不用搞个线程。