zlgopen/awtk #322: list_view动态刷新list_item

webnoise, 2019-12-31:
请问如何高效动态刷新listview中的listitem。
例如搜索到的wifi列表，可能每次扫描出来是动态变化的，假如第一次搜索出来有100个wifi热点，第二次扫描后有105个wifi热点(不一定包含前100个，且顺序是错乱的)。是否需要把listview中的100个wifi热点通过widget_remove_child函数删除掉之后，再通过widget_clone把二次扫描的105个逐个添加？

xianjimli, 2020-01-01:
先调整list item的个数，多就销毁少就创建，再遍历一下，把新的数据设置进去。
你可以参考：https://github.com/zlgopen/awtk-mvvm/blob/master/src/mvvm/awtk/binding_context_awtk.c

```c
static ret_t binding_context_prepare_children(binding_context_t* ctx, widget_t* widget) {
  uint32_t i = 0;
  view_model_t* view_model = ctx->view_model;
  widget_t* template_widget = WIDGET(ctx->template_widget);
  uint32_t items = object_get_prop_int(OBJECT(view_model), VIEW_MODEL_PROP_ITEMS, 0);

  if (ctx->template_widget == NULL) {
    template_widget = widget_get_child(widget, 0);

    ctx->template_widget = template_widget;
    widget_remove_child(widget, template_widget);
  }

  widget_trim_children(widget, items);

  for (i = widget_count_children(widget); i < items; i++) {
    widget_clone(template_widget, widget);
  }
  return_value_if_fail(items == widget_count_children(widget), RET_OOM);

  return RET_OK;
}
```

webnoise, 2020-01-01:
好的，感谢，我试试。

closed 2020-01-01 by webnoise

YinWenshuo, 2020-03-23:
你好，能把你如何使用awtk的这个列表分享出来吗？我最近也要用到list，跟你应用场景差不多，不过我是列出类似于音乐播放列表。可以的话加下我微信：M0oYin