zlgopen/awtk #323: 关于多语言

ytchw101, 2020-01-02:
请问下我现在程序要实现界面显示繁体， 除了String资源中翻译一下， 对应程序也需要加载繁体字体吗？

<string name="OK">
<language name="en_US">OK</language>
<language name="zh_CN">确定</language>
<language name="zh_TD">確定</language>
</string>

注：目前程序使用了一个方正简体(字体资源是编译到程序中的)，

ytchw101, 2020-01-02:
试了下  是要加载繁体字体  ， 但是除了默认的字体是编译到程序中的 ，切换的翻译字体是通过SD卡加载的 ，加载后调用		   
                        LoadFont("Traditional");
		 	 locale_info_change(locale_info(), "zh", "TW");
		 	 system_info_set_default_font(system_info(), "Traditional");
界面上有的繁体能显示有的就显示不出来，，  多操作几次LCD屏界面切换后， 又能显示出来...

xianjimli, 2020-01-02:
要把字体缓存清除掉 明天我看看

ytchw101, 2020-01-03:
目前测试情况是点击按键切换到繁体，有的繁体字出不来。 但是一切换到另外一个特定界面再返回就全变成繁体字了， 切换到其他界面都不行。   那每次我就先显示这个界面在删除，勉强可用但不知道原因。  字体缓存加载之前我也清空 也一样

ytchw101, 2020-01-03:
？？ 加载繁体字.ttf文件，翻译后能显示。  为什么加载个韩文字体(电脑上可显示韩文)， 翻译后不显示韩文？ 和加载繁体字同样的操作。   strings中我只翻译了几个按钮文字翻译成韩文。    有什么要注意的吗》？ 是要把字符串中全部翻译完吗？

ytchw101, 2020-01-03:
除了繁体按照上面步骤可以显示， 我又加了俄文 韩文、 西班牙等  步骤和加繁体步骤一样但是基本都是显示个框框    不知咋办了？

	 switch(lunum)
	 {
	 case 1: //英文
		 locale_info_change(locale_info(), "en", "US");rt_kprintf("en\r\n");
		 break;
	 case 2: /俄文
		 rt_kprintf("rn\r\n");
		 ReleaseFont();
		 LoadFont("CYRILIC");
		 locale_info_change(locale_info(), "rn", "RS");
		 system_info_set_default_font(system_info(), "CYRILIC");
		 break;
	 case 4: //西班牙文
		 rt_kprintf("sh\r\n");
	 	 ReleaseFont();
		 LoadFont("cambriab");
		 locale_info_change(locale_info(), "sh", "SP");
		 system_info_set_default_font(system_info(), "cambriab");
		 break;
	 case 5: //韩文
			 ReleaseFont();
		   LoadFont("korean");
		 	 locale_info_change(locale_info(), "ko", "KR");
		 	 system_info_set_default_font(system_info(), "korean");rt_kprintf("korean\r\n");
			 break;
	 case 6:  //繁体台湾
			 ReleaseFont();
		   LoadFont("Traditional");
		 	 locale_info_change(locale_info(), "zh", "TD");
		 	 system_info_set_default_font(system_info(), "Traditional");rt_kprintf("Traditional\r\n");
			 break;
		default:
			break;
	 }

xianjimli, 2020-01-03:
别着急，你先忙其它事情吧，今天太忙了，明天帮你搞定。

ytchw101, 2020-01-03:
好的 谢谢！

xianjimli, 2020-01-03:
我这里可以了，你更新一下AWTK，参考：https://github.com/zlgopen/awtk-hello

https://github.com/zlgopen/awtk-hello/blob/master/src/settings.c

ytchw101, 2020-01-04:
代码更新到最新(20200103) 第一次上电默认使用内置字体（即assets_manager_add(rm, font_msyh); 这种方式的字体） 界面所有字符串都不显示（注:awtk源码都重新编译，并且把所有资源都重新生成）。

2.之前在界面搞了个按键， 按一次切换一个语言，总共搞了7种语言， 之前的程序切换只有繁体能切换出来，  现在是 有的还是切换不出来都是框框，有的能切换出来 比如韩文， 但是界面有的待翻译的字符串还是之前的语言？

xml strings格式如下
<string name="Chinese">
<language name="en_US">Chinese</language>
<language name="zh_CN">中文</language>
<language name="zh_TD">中文</language>
<language name="rn_RS">китайский язык</language>     
<language name="ko_KR">중국어</language>       
<language name="po_PG">Chinês</language>    
<language name="sh_SP">Chino</language>       
</string>

除了 zh_CN使用内置的font_msyh字体， 其他都是加载SD卡里的.ttf字体

另外更新到最新源码后。之前反应的一个问题就是	int wr_size = fwrite(pread, 1, size, fp_sd);
只能写4096字节问题，又解决了， 测试读写可以

ytchw101, 2020-01-04:
[image]

xianjimli, 2020-01-04:
看看hello正常吗：https://github.com/zlgopen/awtk-hello

ytchw101, 2020-01-04:
[image]

xianjimli, 2020-01-04:
我再awtk-hello中加了韩文测试，也没有问题啊。

xianjimli, 2020-01-04:
感觉这个应该是字体的问题。

xianjimli, 2020-01-04:
先弄个简单的例子试试，在PC上可以运行，如果有问题发给我看看。否则我也没法帮你。

ytchw101, 2020-01-04:
好

ytchw101, 2020-01-04:
你好 把我的代码移到上位机上   测试结果：
1、繁体字我是自己找了个4.5M左右的字体，显示第一遍繁体字出不来，切换下特定界面再返回又出来了。 但是后来我换成大一点的比如13M左右的繁体字体基本上都正常了。

其他的字体有的显示框框应该还是我 找的字体的问题

ytchw101, 2020-01-04:
谢谢了

ytchw101, 2020-01-09:
中文 韩文(约6.3M)  俄文 西班牙文 葡萄牙文 （约1M）  除了中文都是从SD卡加载， 语言切换都正常。
但是繁体字 找个4.2M左右的简繁体，加载后繁体字部分就是翻译不过来。 换成大一点的比如14M左右的繁体字体就可以（但太耗空间）！    这个简繁体在程序中使用解析某个字比如  “时间” ,解析出来是繁体字 “時間”，

xianjimli, 2020-01-09:
你可以提取常用字出来试试：https://blog.csdn.net/linxinfa/article/details/88427808

ytchw101, 2020-01-09:
谢谢  我来试下

WeSrc-Yin, 2021-09-28:
兄弟，你字库哪里下载的，我网上找字库，怎么全是艺术字库，没有那种常规字库。
@ytchw101

ytchw101, 2021-10-09:
我这有几十个自己下载的矢量ttf字体  需要可以打包发你