zlgopen/awtk #324: 更新最新源码后问题

ytchw101, 2020-01-02:
之前程序使用一直正常， 更新到最新AWTK源码(2019/12/31)后，

1. 多行编辑器界面当编辑多个汉字比如100个，点击确定按钮，需要点击两次才进入响应函数。
2.还有个情况是有的情况输入法键盘关闭不了。像我目前的程序更新到最新就有，
（简单叙述：一个界面创建个edit控件，同时设置为focus,键盘弹， 界面一个按钮点击后打开一个ui在界面显示， 点击ui界面的关闭按钮,ui界面关闭， 同时再次设置界面的edit为focus 键盘弹起， 当点击界面关闭按钮时， 界面是关了 但是输入法键盘没关？）
3.关于显示二维码的部分有更新吗，为什么现在显示二维码崩溃？

xianjimli, 2020-01-02:
1和2明天我看看；3我不清楚、你发二维码的仓库上吧

ytchw101, 2020-01-03:
4.还有个问题就是，更新源码之前的代码文件操作读写U盘SD卡正常的 ，现在 只能写4096字节就不行了 不知为啥：主要代码
		int size = fread(pread, 1, read_size, fp_u); //从U盘读内容
		if(size == 0)
		{
			iserr = TRUE;
			break;
		}			

		int wr_size = fwrite(pread, 1, size, fp_sd);//写到SD卡
		
		if(wr_size != size) 
		{
			rt_kprintf("\r\n read and write size is not same!\r\n");
			iserr = TRUE;
			break;
		}

之前测试一直没问题， 更新源码后只能写到4096字节 ， 就提示wr_size  = 0了  ， fs_oc.c还是用的之前写好测试可以的，没变过。。。

WNsACE, 2020-01-03:
能不能把第一，二问题的xml发出来呢，比较容易从xml中分析出问题，可以做一个简单的存在上述问题的xml发出来，让我们分析