zlgopen/awtk #326: 我使用rich_text的过程中text_align_h没有生效

gaoyuan1111, 2020-01-07:
我在使用rich_text的过程中，发现text_align_h并没有生效。无论是right还是center。请问是我哪里写错了吗，还是什么其它的原因。 我的目的就是为了实现字符串的居中，并且自动换行后也要居中。下面是我实现的代码。

[image]

[image]
[image]

xianjimli, 2020-01-07:
text_align_h还没实现。近期加上吧。

gaoyuan1111, 2020-01-07:
我刚在使用中发现字体大小也是不生效的。这个也是没有实现的吗？ 还是我使用方法出错了

xianjimli, 2020-01-07:
字体大小在font属性里指定，看下demoui

xianjimli, 2020-01-08:
支持了text_align_h，你更新代码试试。

closed 2020-03-13 by xianjimli

scp1011, 2023-04-21:
> 支持了text_align_h，你更新代码试试。

text_align_v会支持吗

xianjimli, 2023-04-23:
> text_align_v会支持吗

我记得是支持的。

scp1011, 2023-04-24:
> > text_align_v会支持吗
> 
> 我记得是支持的。

好的，但我在designer里是没有实时看到效果，所以就没跑代码看效果