zlgopen/awtk #339: 如何裁剪谷歌输入法中的中文字词

lizhenningi, 2020-02-05:
您好，新年快乐！
我现在使用AWTK自带的谷歌中文输入法，但是我使用的字库中支持的汉字不够多，因此在候选框中显示搜索到的汉字有的显示不出来，我想把这一部分显示不出来的汉字去掉，这个要怎么弄？

xianjimli, 2020-02-06:
你更新一下代码，参考这个文档：

https://github.com/zlgopen/awtk/blob/master/docs/how_to_update_gpinyin_data.md

lizhenningi, 2020-02-06:
好的，我试一下，谢谢您

lizhenningi, 2020-02-06:
您好！
我更新AWTK代码，在Windows上使用scons编译不通过。
我将gpinyin_gen.cpp中的#include <unistd.h>改为#include <io.h>与#include <process.h>，之后再次编译出现如下图所示错误，我的c++功底较浅，望您指教！
[image]

xianjimli, 2020-02-06:
你再更新一下。谢谢

lizhenningi, 2020-02-06:
好的，谢谢您

lizhenningi, 2020-02-06:
您好！
我想将谷歌拼音输入法中的联想功能去掉，要怎么做

lizhenningi, 2020-02-06:
您好！
我当前的项目中使用的AWTK代码的版本较低，我不想更新项目中的AWTK代码，
但是我使用最新AWTK代码生成新的字典文件，这个新的字典文件放到我项目中对应的文件夹下，这样能生效吗？这个新的字典文件是否是3rd/gpinyin/data/gpinyin.dat

xianjimli, 2020-02-06:
我增加了宏：WITHOUT_SUGGEST_WORDS 用于关闭联想功能。

https://github.com/zlgopen/awtk/blob/master/src/base/awtk_config_sample.h

```c
/**
 * 启用输入法，但不想启用联想功能，请定义本宏。
 *
 * #define WITHOUT_SUGGEST_WORDS 1
 */
```

xianjimli, 2020-02-06:
目前的版本比较稳定，你可以更新到最新代码。

lizhenningi, 2020-02-06:
您好！
是不是说将最新AWTK代码生成新的字典文件，放到低版本的awtk中是不生效的？

xianjimli, 2020-02-06:
应该是可以的。变量名需要修改一下。

closed 2020-03-13 by xianjimli