zlgopen/awtk #341: 使用awtk-linux-fb成功移植，但是button有点问题想请教

bill611, 2020-02-10:
您好，我成功移植到了arm linux平台（芯唐n32926），无鼠标，只有触摸屏。
当直接触摸按下按钮时，按钮响应了EVT_POINTER_DOWN事件，但是图标状态却没有变化，
当手指抬起时，按钮相应了EVT_POINTER_UP事件，图标状态变成了press状态，
再点击屏幕任意地方，按钮失去焦点，图标变成了normal状态。
在windows和ubuntu模拟都是正常的，因为有鼠标的原因，按钮都会提前进入over状态。
请问是我移植的时候哪里配置不对吗？

xianjimli, 2020-02-10:
一般是上报的触屏事件不对，你把上报的触屏事件打出来看看。

bill611, 2020-02-11:
当手指按下的时候是上报down,抬起上报Up。没有打印EVT_POINTER_的是触摸在按钮外区域，打印了EVT_POINTER_的是触摸在按钮上。
tslib打印是在awtk-linux-fb/awtk-port/input_dispatcher.c的input_dispatch_print中。
EVT_POINTER_打印是在awtk/src/widgets/button.c 的button_on_event函数的case中
/tmp1/release # ./bin/demo
try /tmp1/release
app_root=/tmp1/release
fb_info_t: /dev/fb0
xres=800 yres=480
xres_virtual=800 yres_virtual=960
bits_per_pixel=16 line_length=1600
fb_info_t: red(11 5) green(5 6) blue(0 5)
xpanstep=0 ywrapstep=0
fb_size=768000 fb_total_size=2304000 fb_nr=2 smem_len=2304000
fb_open clear
fb_open ok
ratio=1.000000 800 480
Loading module /usr/gui/tslib/input.so
Loading module /usr/gui/tslib/pthres.so
Loading module /usr/gui/tslib/variance.so
Loading module /usr/gui/tslib/dejitter.so
Loading module /usr/gui/tslib/linear.so
used: 0 bytes 167 blocks
window home_page open
HAVE FD_CLOEXEC
tslib pointer down:315 309
tslib pointer up:315 309
tslib pointer down:366 214
EVT_POINTER_DOWN
tslib pointer up:366 214
EVT_POINTER_UP
tslib pointer down:375 219
EVT_POINTER_DOWN
tslib pointer up:375 219
EVT_POINTER_UP
tslib pointer down:370 223
EVT_POINTER_DOWN
tslib pointer up:370 223
EVT_POINTER_UP
tslib pointer down:366 217
EVT_POINTER_DOWN
tslib pointer up:366 217
EVT_POINTER_UP

bill611, 2020-02-11:
对了，我修改了awtk-port/tslib_thread.c里对ret = ts_read(info->ts, &e, 1);返回值ret的判断
当ret==0时，也算成功，否则每触摸一下，tslib就会重新加载
diff --git a/awtk-port/tslib_thread.c b/awtk-port/tslib_thread.c
index a594c28..cb053de 100644
--- a/awtk-port/tslib_thread.c
+++ b/awtk-port/tslib_thread.c
@@ -1,4 +1,4 @@
-<U+FEFF>/**
+/**
  * File:   tslib_thread.c
  * Author: AWTK Develop Team
  * Brief:  thread to handle touch screen events
@@ -57,7 +57,7 @@ static ret_t tslib_dispatch_one_event(run_info_t* info) {
 
   event_queue_req_t* req = &(info->req);
 
-  if (ret <= 0) {
+  if (ret < 0) {
     printf("%s:%d tslib read failed(ret=%d, errno=%d, filename=%s)\n", __func__, __LINE__, ret,
            errno, info->filename);
     perror("Print tslib: ");
@@ -79,8 +79,8 @@ static ret_t tslib_dispatch_one_event(run_info_t* info) {
   req->pointer_event.x = e.x;
   req->pointer_event.y = e.y;

bill611, 2020-02-12:
找了几天终于找到原因了，原来是lcd_mem_linux_begin_frame里判断起始fb不一样。怪不得显示的状态和实际的状态是反的，动画什么都不流畅。 现在完全流畅了。
另外在awtk-port/tslib_thread.c中修改了EVT_POINTER_MOVE上报的时机，按住没动的话不上报move
[image]