zlgopen/awtk #344: 关于awtk在linux平台上的触摸问题 请教一下

YinWenshuo, 2020-02-18:
要支持触摸，我做了下面几项工作：（arm-linux平台）
1、配置awtk-linux-fb的awtk_config.py文件里面TSLIB_LIB_DIR=‘xxxxxxxxxx’  /*使用我所使用的芯片的编译链编译出的tslib放在这个目录下*/
2、#define TS_DEVICE_FILENAME  "/dev/input/event0"

触摸工作还是不正常，我的问题是：
1、还有哪里没有配置到吗？
2、github上tslib源码太多了，我下了一份，然后交叉编译出库文件。但不知道这个tslib源码是不是正常的，awtk的兄弟能发个tslib源码的链接给我吗？
3、awtk有没有触摸配置、使用相关的比较详细的文档？

谢谢！！！

YinWenshuo, 2020-02-19:
@xianjimli 有空帮忙解答下，谢谢

xianjimli, 2020-02-19:
不好意思，我对tslib没有研究，只是用了公司板子提供的库。

YinWenshuo, 2020-02-19:
> 不好意思，我对tslib没有研究，只是用了公司板子提供的库。

那我在awtk里面那样调试触摸，步骤没错吧？我说的是awtk这边的步骤，如果步骤没错，那么我就去排查tslib方面的问题。

xianjimli, 2020-02-19:
1.先用tslib自带的测试程序，看看读取的点是否正常。
2.再看awtk调用tslib读取的点是否正常。

YinWenshuo, 2020-02-20:
> 1.先用tslib自带的测试程序，看看读取的点是否正常。
> 2.再看awtk调用tslib读取的点是否正常。

好的，我先做第一步，如果第一步正常，第二步不正常再来请教你。
谢谢。

YinWenshuo, 2020-02-20:
李老师，我用tslib的测试程序，测试是正常的，左上角坐标（0,0）；右下角坐标最大（800,720），按下标记255，抬起标记0。在awtk里面跑起来，触摸操作，打印的是mouse pointer down，而且坐标对不上，偏移很大。

YinWenshuo, 2020-02-20:
可以了，谢谢。

YinWenshuo, 2020-02-20:
触摸打印的坐标是对的，但是跑完触摸的代码怎么还会跑鼠标的代码？我把鼠标关掉，触摸完全正常。
鼠标跟触摸不能同时？

closed 2020-02-22 by YinWenshuo