zlgopen/awtk #346: 嵌入式环境颜色失真

xinna, 2020-02-19:
我使用AWTK Designer设计的界面。 编译后下载到主板上颜色全部失真，图片、控件颜色也都不一样。
还有圆角也变成直角了，第一次移植，请教问题在哪里？
AWTK Designer设计图
链接：https://pan.baidu.com/s/1TeBPB_qM2315un3i75YjsQ 
提取码：ro02 
复制这段内容后打开百度网盘手机App，操作更方便哦

显示出来的图
链接：https://pan.baidu.com/s/1fBLK26muoY8ZRUoMYtxiRQ 
提取码：ct13 

原始图片
链接：https://pan.baidu.com/s/1-QUjra1RDBFfqk39sG_saQ 
提取码：8dxi 
复制这段内容后打开百度网盘手机App，操作更方便哦

我是用图片控件直接显示的，显示过原始的jpg文件，也用image2lcd转换成16bit的bmp。都显示一样。
裸机环境 F1C100S   master分支，Keil编译。

xianjimli, 2020-02-19:
用lcd_mem_bgr565_create_three_fb试试。

xinna, 2020-02-20:
> 用lcd_mem_bgr565_create_three_fb试试。

确实是这个问题，昨晚吧datasheet又看了一边，也在怀疑字节序问题。
这个问题解决了。

但是圆角的为啥都画不出来呢。

WNsACE, 2020-02-20:
不单单是圆角没有画出来，还有右边的圆形进度条也没有画出来，感觉像是 vg 没有创建出来的样子，可以检查一下 lcd_mem.inc 文件中的 lcd_mem_get_vgcanvas 函数中的 vgcanvas_create 是否正常调用。

xinna, 2020-02-20:
> 不单单是圆角没有画出来，还有右边的圆形进度条也没有画出来，感觉像是 vg 没有创建出来的样子，可以检查一下 lcd_mem.inc 文件中的 lcd_mem_get_vgcanvas 函数中的 vgcanvas_create 是否正常调用。

是vg没有创建。在master上vg我没编译过去，错误是找不到vgcanvas_nanovg_fbo_to_bitmap
在vgcanvas_nanovg.inc里面。我全文件夹搜索找不到这个方法的定义。

无奈下载1.3.1分支，全部重新移植，加上vg，圆角、圆形进度条 都可以了。

closed 2020-02-20 by xinna