zlgopen/awtk #348: 全志F1c200s运行官方Tina linux系统，移植AWTK，闪烁严重，请教怎么解决

duanchangstar, 2020-02-23:
全志F1c200s运行官方Tina linux系统，运行自带的minigui和播放mp4视频正常。
视频和移植的AWTK程序都在SD卡上，编译AWTK播放器的Demo
闪烁非常严重。请教怎么解决。

视频上传到了B站
https://www.bilibili.com/video/av91122136/

打印信息如下：
`root@TinaLinux:/mnt/SDCARD/ha/bin# ./demo
try /mnt/SDCARD/ha/bin
app_root=/mnt/SDCARD/ha
fb_info_t: /dev/fb0
xres=800 yres=480
xres_virtual=800 yres_virtual=960
bits_per_pixel=32 line_length=3200
fb_info_t: red(16 8) green(8 8) blue(0 8)
xpanstep=1 ywrapstep=0
fb_size=1536000 fb_total_size=3072000 fb_nr=2 smem_len=3072000
fb_open clear
fb_open ok
ratio=1.000000 800 480
!!!Asset [name=yellow type=data] not exist!!!
used: 0 bytes 785 blocks
window main open
FBIOPAN_DISPLAY ret=0 yoffset=0
FBIOPAN_DISPLAY ret=0 yoffset=480
FBIOPAN_DISPLAY ret=0 yoffset=0
FBIOPAN_DISPLAY ret=0 yoffset=480
FBIOPAN_DISPLAY ret=0 yoffset=0
FBIOPAN_DISPLAY ret=0 yoffset=480
FBIOPAN_DISPLAY ret=0 yoffset=0
FBIOPAN_DISPLAY ret=0 yoffset=480
FBIOPAN_DISPLAY ret=0 yoffset=0`

xianjimli, 2020-02-24:
貌似驱动没有正确实现FBIOPAN_DISPLAY，你用flush的方式试试。

```c
static lcd_t* lcd_linux_create(fb_info_t* fb) {
    return lcd_linux_create_flushable(fb);
}
```

duanchangstar, 2020-03-10:
> 貌似驱动没有正确实现FBIOPAN_DISPLAY，你用flush的方式试试。
> 
> ```c
> static lcd_t* lcd_linux_create(fb_info_t* fb) {
>     return lcd_linux_create_flushable(fb);
> }
> ```

已解决，谢谢

closed 2020-04-11 by xianjimli