zlgopen/awtk #355: 请问有没有移植文档，针对g2d硬件加速的？

cgha, 2020-03-06:
touchgfx有DMA移植的例子，awtk我看了一些移植还是没概念该怎么开启硬件加速？我看有针对ST的硬件加速，哪个工程是？

xianjimli, 2020-03-06:
定义WITH_STM32_G2D即可：
https://github.com/zlgopen/awtk/blob/master/src/blend/stm32_g2d.c

xianjimli, 2020-03-06:
https://github.com/zlgopen/awtk-stm32f429igtx-raw

cgha, 2020-03-07:
好的，我试着移植一下到我的板子上(keil平台下SDK自带的例子)，好多错误不知道怎么解决，可能是--gnu参数导致的。现在一般都是拿官方SDK来改，官方SDK一般针对各个平台有宏适配，加入--gnu后导致跟其中一些原有的宏定义冲突，我觉得是否能针对平台再分细一点，比如arm_cc, icc，不然的话新手移植还是比较难。

xianjimli, 2020-03-07:
我们的人力有限，欢迎大家一起来完善。

cgha, 2020-03-09:
请问AWTK是怎么跟LCD硬件联系到一起的，比如：MCU有自己的TFT控制器，或者用的ST7789之类的专用控制器。在我脑子里的概念是要调用set_pixel，get_pixel, draw_vline, draw_hline之类的函数，或者使用fb的话就是调用render_fb之类的函数。我看了有一个lcd.inc，刚开始创建fb的时候返回lcd_t*，但是好像看了例子并没有往里边填各种具体操作的函数？

xianjimli, 2020-03-09:
请参考：https://github.com/zlgopen/awtk/blob/master/docs/lcd.md