zlgopen/awtk #357: 使用window manager响应esc按键后,esc按下的事件进不能进入控件的on_key_down事件回调了

21pages, 2020-03-09:
widget_t* wm = window_manager();
 widget_on(wm, EVT_KEY_DOWN, on_key_back, wm);

static ret_t on_key_back(void* ctx, event_t* e) {
  key_event_t* evt = (key_event_t*)e;
  widget_t *top = window_manager_get_top_window(WIDGET(ctx));
  if (evt->key == TK_KEY_ESCAPE) {
    if(top && tk_str_eq(top->style, "from_pwd")) {
      window_manager_back(WIDGET(ctx));
      return RET_OK;
    } else {
      window_manager_back(WIDGET(ctx));
      return RET_STOP;
    }
  }
  return RET_OK;
}
esc是用来通用退出界面的,但是自定义了一个空间,esc用来删除字符,对应的是面板上的取消键,这个该怎么处理?

xianjimli, 2020-03-09:
你是想让控件优先处理，控件不处理再让窗口处理器处理吗？

21pages, 2020-03-09:
是的

xianjimli, 2020-03-09:
控件处理之后返回 RET_STOP 就行了。

21pages, 2020-03-09:
并不会进控件的按键处理函数

xianjimli, 2020-03-09:
明天我看看

21pages, 2020-03-10:
我想到一个办法,当控件进入编辑态的时候,使用widget_off注销window manger对于按键的相应,退出编辑态后重新注册,但这有点麻烦, 有更好的解决办法吗

21pages, 2020-03-10:
我用上面说的方法,基本实现了,但是重新注册window manager按键相应后还是对刚刚的esc相应了,用event_destroy删除该事件会报错,用RET_STOP也不行,该怎么不让该事件继续传递

21pages, 2020-03-10:
用上面的方法解决了

closed 2020-03-10 by 21pages