zlgopen/awtk #358: 为什么输入框是弹出个键盘？

243011068, 2020-03-10:
win系统不需要弹键盘啊，而且中文也输入不进去

xianjimli, 2020-03-10:
app type 设置为：APP_DESKTOP

请参考：demos/demo_desktop.c

243011068, 2020-03-11:
我看了下是define个值，nodejs中如何定义？
[image]
而且这个init第三个参数是type，第四个才是标题，源文件有误，type我传的TAppType.DESKTOP也无效

xianjimli, 2020-03-11:
nodejs 版本，编译 PC 版本，请把 src/c/custom.c 中的APP\_SIMULATOR 改成 APP\_DESKTOP，并重新编译。

243011068, 2020-03-12:
[image]
可以正常输入了，但是有个bug文字输入进去之后，输入上的字还在

xianjimli, 2020-03-12:
这个是 Windows 和 SDL 的交互，不同平台有些差异，AWTK 没法控制。

243011068, 2020-03-12:
没什么办法么，这个太影响体验了

xianjimli, 2020-03-12:
什么平台？得研究一下 SDL。

243011068, 2020-03-12:
win7,64位

xianjimli, 2020-03-13:
我们测试了几台机器，没有重现。

243011068, 2020-03-13:
可能还有什么条件没满足吧，
[image]
zh'ge这个为什么不直接读一下第三个参数？

closed 2020-03-18 by 243011068