zlgopen/awtk #361: scrollview下弹出键盘,scrollview会自动滑动到最底部,原本的控件是在scrollview上方的话,就看不到了

21pages, 2020-03-12:
rt

xianjimli, 2020-03-12:
把 xml 贴上来看看。

21pages, 2020-03-12:
屏幕是160*160的,有上下两个systembar各16,主屏128高度,下面那个额定电流的edit,是最简单的,但是有上面两个问题,我可以加您微信吗,可以给您发gif

<window move_focus_prev_key="left" move_focus_next_key="right">
  <list_view w="100%" h="100%" item_height="21" auto_hide_scroll_bar="false"
             focused="true" map_key:up="pageup" map_key:down="pagedown">
    <scroll_view name="view" w="100%" h="100%" children_layout="list_view(xm=4)">
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="配置序号"/>
        <edit_num  name="edit_num_index" w="60%" h="100%" hex="false" fixed_length="false" max_length="5" focusable="true" map_key:pageup="up" map_key:pagedown="down" select_none_when_focused="true"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="表地址"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <edit_num name="edit_num_tsa_add" w="100%" h="100%" hex="false" fixed_length="false" max_length="12" focusable="true" map_key:pageup="up" map_key:pagedown="down" select_none_when_focused="true"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="逻辑地址"/>
        <spin_box name="spin_box_tsa_vxd" w="60%" h="100%" input_type="uint" min="0" max="3" value="0" focusable="true" map_key:pageup="up" map_key:pagedown="down" open_im_when_focused="false" auto_fix="true"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="地址类型"/>
        <combo_box name="combo_box_tsa_type" readonly="true" w="60%" h="100%" focusable="true"
          map_key:up="1" map_key:down="1" map_key:return="1" map_key:pageup="up" map_key:pagedown="down"
          options="单地址;通配地址;组地址;广播地址" selected_index="0">
          <button style="combobox_down" x="right" y="center" w="10" h="100%"/>
        </combo_box>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="波特率"/>
        <combo_box name="combo_box_baud" readonly="true" w="60%" h="100%" focusable="true"
          map_key:up="1" map_key:down="1" map_key:return="1" map_key:pageup="up" map_key:pagedown="down"
          options="300;600;1200;2400;4800;7200;9600;19200;38400;57600;115200;255:自适应;" selected_index="0">
          <button style="combobox_down" x="right" y="center" w="10" h="100%"/>
        </combo_box>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="通讯规约"/>
      </list_item>
      <list_item>
        <combo_box name="combo_box_protocol" readonly="true" w="100%" h="100%" focusable="true"
          map_key:up="1" map_key:down="1" map_key:return="1" map_key:pageup="up" map_key:pagedown="down"
          selected_index="0" options="未知规约;DLT645_97;DLT645_07;DLT698_45;DLT_188;无效规约">
          <button style="combobox_down" x="right" y="center" w="10" h="100%"/>
        </combo_box>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="通讯端口"/>
        <combo_box name="combo_box_port" readonly="true" w="60%" h="100%" focusable="true"
          map_key:up="1" map_key:down="1" map_key:return="1" map_key:pageup="up" map_key:pagedown="down"
          options="交采端口;1#485;2#485;载波无线" selected_index="0">
          <button style="combobox_down" x="right" y="center" w="10" h="100%"/>
        </combo_box>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="通信密码"/>
        <edit_num name="edit_num_password" w="60%" h="100%" hex="true" hex_lower="false" fixed_length="false" max_length="8" focusable="true" map_key:pageup="up" map_key:pagedown="down"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="费率个数"/>
        <spin_box name="spin_box_rateNum" w="60%" h="100%" input_type="uint" min="0" max="255" auto_fix="true" focusable="true" select_none_when_focused="true" open_im_when_focused="false" map_key:pageup="up" map_key:pagedown="down" map_key:return="tab"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="用户类型"/>
        <edit_num name="edit_num_userType" w="60%" h="100%" hex="false" fixed_length="false" max_length="5" focusable="true" map_key:pageup="up" map_key:pagedown="down"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="接线方式"/>
        <combo_box name="combo_box_pwrType" readonly="true" w="60%" h="100%" focusable="true"
          map_key:up="1" map_key:down="1" map_key:return="1" map_key:pageup="up" map_key:pagedown="down"
[... 4 lines cut ...]
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="额定电压"/>
        
        <edit_num name="edit_num_stVolt" w="60%" h="100%" hex="false" fixed_length="true" text="000.0" focusable="true" map_key:pageup="up" map_key:pagedown="down"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="额定电流"/>
        <edit name="edit_stAmp" w="60%" h="100%" input_type="ufloat" map_key:pageup="up" map_key:pagedown="down"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <label w="40%" style="left" text="采集器"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
[... cut ...] ="up" map_key:pagedown="down"/>
      </list_item>
      <list_item children_layout="default(r=1,c=0)">
        <column w="100%" h="100%" children_layout="default(r=1,c=2)">
          <button name="func:ui_meter_add" border_color="black" focusable="true" text="确定"/>
          <button name="close" border_color="black" focusable="true" text="取消"/>
        </column>
      </list_item>
    </scroll_view>
    <scroll_bar_m name="bar" x="right" y="0" w="3" h="100%" value="0"/>
  </list_view>
</window>

xianjimli, 2020-03-12:
> scrollview会自动滑动到最顶部 

这个问题在我这里没出现，你用最新代码试试。

> 设好input_type后,demoui里弹出的键盘是不同的,而我这里只弹出了一个编辑text的

把 edit_num 换成 edit 时，键盘是正常的，我要看看自定义控件 edit_num 实现才知道。

21pages, 2020-03-13:
异常的控件是额定电流所在的那个list_item, 是edit

21pages, 2020-03-13:
新的master程序也是只有一种软键盘,不知道我哪里需要设置一下

xianjimli, 2020-03-13:
kb_xxxx资源存在吗

21pages, 2020-03-13:
` <keyboard grab_keys="true" theme="keyboard" x="0" y="bottom" w="100%" h="50%" move_focus_up_key="up"
   move_focus_down_key="down" move_focus_left_key="left" move_focus_right_key="right">
    <pages x="0" y="bottom" w="100%" h="100%" active="0">
      <view name="upper" x="0" y="0" w="100%" h="100%" children_layout="default(r=4,c=1,s=1,m=1)">
        <group_box children_layout="default(r=1,c=10,s=1,m=1)">
          <button style="kb_ascii" focusable="true" repeat="300" name="Q" text="Q"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="W" text="W"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="E" text="E"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="R" text="R"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="T" text="T"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="Y" text="Y"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="U" text="U"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="I" text="I"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="O" text="O"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="P" text="P"/>
        </group_box>
        <group_box children_layout="default(r=1,c=9,s=1,xm=5,ym=1)">
          <button style="kb_ascii" focusable="true" repeat="300" name="A" text="A"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="S" text="S"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="D" text="D"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="F" text="F"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="G" text="G"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="H" text="H"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="J" text="J"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="K" text="K"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="L" text="L"/>
        </group_box>
        <group_box children_layout="default(r=1,c=9,s=1,m=1)">
          <button style="kb_ascii" focusable="true" name="page:lower"><image draw_type="icon" image="shifton" x="center" y="middle" w="100%" h="100%"/></button>
          <button style="kb_ascii" focusable="true" repeat="300" name="Z" text="Z"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="X" text="X"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="C" text="C"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="V" text="V"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="B" text="B"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="N" text="N"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="M" text="M"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="key:backspace"><image draw_type="icon" image="backspace" x="center" y="middle" w="100%" h="100%"/></button>
        </group_box>
        <group_box children_layout="default(r=1,c=0,s=1,m=1)">
          <button style="kb_ascii" focusable="true" name="page:symnum" w="30%" text="123"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="space" w="40%" text="Space"/>
          <button style="kb_ascii" focusable="true" name="close" w="30%" text="Return"/>
        </group_box>
      </view>

      <view name="lower" x="0" y="0" w="100%" h="100%" children_layout="default(r=4,c=1,s=1,m=1)">
        <group_box children_layout="default(r=1,c=10,s=1,m=1)">
          <button style="kb_ascii" focusable="true" repeat="300" name="q" text="q"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="w" text="w"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="e" text="e"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="r" text="r"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="t" text="t"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="y" text="y"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="u" text="u"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="i" text="i"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="o" text="o"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="p" text="p"/>
        </group_box>
        <group_box children_layout="default(r=1,c=9,s=1,xm=5,ym=1)">
          <button style="kb_ascii" focusable="true" repeat="300" name="a" text="a"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="s" text="s"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="d" text="d"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="f" text="f"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="g" text="g"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="h" text="h"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="j" text="j"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="k" text="k"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="l" text="l"/>
        </group_box>
        <group_box children_layout="default(r=1,c=9,s=1,m=1)">
[... cut ...]
          <button style="kb_ascii" focusable="true" repeat="300" name="key:backspace"><image draw_type="icon" image="backspace" x="center" y="middle" w="100%" h="100%"/></button>
        </group_box>
        <group_box children_layout="default(r=1,c=0,s=1,m=1)">
          <button style="kb_ascii" focusable="true" name="page:upper" w="30%" text="ABC"/>
          <button style="kb_ascii" focusable="true" repeat="300" name="space" w="40%" text="Space"/>
          <button style="kb_ascii" focusable="true" name="close" w="30%" text="Return"/>
        </group_box>
      </view>
    </pages>
</window>`

21pages, 2020-03-13:
我知道了,我没添加到资源

closed 2020-04-06 by 21pages