zlgopen/awtk #374: slide_view里使用软键盘,按左右键选择按键,会导致slide_view切换

21pages, 2020-03-21:
希望键盘弹出时不要切换slide_view,键盘代码如下
`<keyboard grab_keys="true" theme="keyboard" x="0" y="bottom" w="100%" h="50%"
        move_focus_up_key="up" move_focus_down_key="down" move_focus_left_key="left" move_focus_right_key="right">
  <column style="placeholder" x="0" y="0" w="100%" h="10">
    <button name="close" focusable="true" style="close" x="r:1" y="1" w="8" h="8" text="x" />
  </column>
  <pages x="0" y="10" w="100%" h="-10" active="0">
    <view name="upper" x="0" y="0" w="100%" h="100%" children_layout="default(r=4,c=3,s=2,m=4)">
        <button style="kb_ascii" focusable="true" name="1" text="1"/>
        <button style="kb_ascii" focusable="true" name="2" text="2"/>
        <button style="kb_ascii" focusable="true" name="3" text="3"/>
        <button style="kb_ascii" focusable="true" name="4" text="4"/>
        <button style="kb_ascii" focusable="true" name="5" text="5"/>
        <button style="kb_ascii" focusable="true" name="6" text="6"/>
        <button style="kb_ascii" focusable="true" name="7" text="7"/>
        <button style="kb_ascii" focusable="true" name="8" text="8"/>
        <button style="kb_ascii" focusable="true" name="9" text="9"/>
        <button style="kb_ascii" focusable="true" name="-" text="-"/>
        <button style="kb_ascii" focusable="true" name="0" text="0"/>
        <button style="kb_ascii" focusable="true" name="key:backspace">
          <image draw_type="icon" image="backspace" x="center" y="middle" w="100%" h="100%"/>
        </button>
    </view>
  </pages>
</keyboard>`
还有之前提的,键盘弹出时,编辑控件不能显示的问题,我看代码里好像不会调到widget_ensure_visible_in_scroll_view,希望能改下

xianjimli, 2020-03-22:
> 键盘弹出时,编辑控件不能显示的问题,我看代码里好像不会调到widget_ensure_visible_in_scroll_view,希望能改下

什么问题？详细说明一下。

21pages, 2020-03-22:
> > 键盘弹出时,编辑控件不能显示的问题,我看代码里好像不会调到widget_ensure_visible_in_scroll_view,希望能改下
> 
> 什么问题？详细说明一下。

被编辑的控件被顶部systembar遮挡了

21pages, 2020-03-22:
还有一个问题,就是edit_t控件,focused时左右按键是移动光标,文本长就要按好久,能不能有个编辑态,非编辑态下直接移动到相邻控件

21pages, 2020-03-22:
edit_t光标的问题,我解决了,在edit_on_key_down函数开头加
`if (!edit_is_number(widget)) {
		if(!edit->in_edit) {
			if (e->key != TK_KEY_RETURN) {
	      		return RET_OK;
	    	} else {
				edit->in_edit = 1;
				text_edit_set_cursor(edit->model, edit->widget.text.size);
			}
		} else {
			if(key == TK_KEY_LEFT) {
				text_edit_set_cursor(edit->model, 0);
			}
		}
	} else {
			if(key == TK_KEY_LEFT) {
				text_edit_set_cursor(edit->model, 0);
			} else if(key == TK_KEY_RIGHT) {
				text_edit_set_cursor(edit->model, widget->text.size);
			}
	}`

xianjimli, 2020-03-22:
> > > 键盘弹出时,编辑控件不能显示的问题,我看代码里好像不会调到widget_ensure_visible_in_scroll_view,希望能改下
> > 
> > 
> > 什么问题？详细说明一下。
> 
> 被编辑的控件被顶部systembar遮挡了

这个不太好改。对于小屏幕，建议打开一个新窗口进行编辑，用label模拟一个编辑器，按下回车时打开新窗口，新窗口上只有一个编辑器。就像android早期的处理方法一下。

21pages, 2020-03-23:
slide_view,软键盘,左右按键,这个好改吗?

21pages, 2020-03-23:
用新代码解决了

closed 2020-03-23 by 21pages