zlgopen/awtk #377: widget_set_text,  msvc汉字编码问题

21pages, 2020-03-30:
程序文件编码:utf8,出现
问题1:汉字是奇数个且没有标点时,报错error C2001: 常量中有换行符
问题2:汉字偶数个时才能正确显示

方法1:
widget_set_text_utf8(widget, "有错误!"), 或widget_set_prop_str(widget, WIDGET_PROP_STR, "有错误!"),
因为带了叹号,所以能编译通过,但是只能显示"有错"
方法2:
char * s = "有错误!"
wchar_t ws[32];
tk_utf8_to_utf16(s, ws, 32);
widget_set_text(label, ws); 乱码显示 
方法3:
wchar_t *ws = L"有错误!";
widget_set_text(label);
我看有些代码这么写可以,  但是我这么写还是不行

我想问到底该怎么写?

xianjimli, 2020-03-30:
文件格式用UTF8-BOM，MSVC才会把它当作UTF8格式。

21pages, 2020-03-30:
我重新用ut8-bom保存,结果变成乱码了,显示不出来了

xianjimli, 2020-03-30:
用vscode，以UTF-8重新打开，再保存UTF8-bom。MSVC打开的时候已经错了，保存自然是错的。

xianjimli, 2020-03-30:
中文最好放到locales里，代码里都用英文。

21pages, 2020-03-30:
我就是用的vscode...用notepad++也不行,有没有个一定行,通用的办法

xianjimli, 2020-03-30:
那我不知道了。把中文写到locales的XML文件中吧。

21pages, 2020-03-31:
是的,采用的方案是1.文件编码用utf8-bom, 2.字符串前加L ,类型为wchar_t* 3.使用widget_set_text,就可以了

closed 2020-03-31 by 21pages